tinygo-org/tinygo #3279: Interrupts with Arduino Uno

boisgera, 2022-11-08:
Hi everyone! 

I am using TinyGo for Arduino Uno and I am having a great time! Thanks to everyone involved in this project 👍

The only hurdle I have so far is the management of GPIO interrupts. I am following the doc [here](https://tinygo.org/docs/concepts/peripherals/gpio/#interrupts-on-pin-changes) ; but Pin seems to have no `SetInterrupt` method if I am to believe the compiler (tinygo v0.26) or VS Code ([properly configured](https://tinygo.org/docs/guides/ide-integration/vscode/) AFAICT). Am I missing something obvious? 🤔

Regards,

Sébastien

aykevl, 2022-11-17:
Unfortunately, `SetInterrupt` has not yet been implemented on AVR chips. The hardware supports it so it's just a matter of implementing it.

boisgera, 2022-11-17:
> Unfortunately, `SetInterrupt` has not yet been implemented on AVR chips. The hardware supports it so it's just a matter of implementing it.

Ah, OK, good to know! I am afraid that I am not familiar enough with TinyGo internals to be of any help here.
Thank you for the reply!

BerndCzech, 2022-12-26:
I would be willing to have a look as I have an Uno here.

deadprogram, 2023-05-18:
Duplicate of #3145 so now closing.