tinygo-org/tinygo #3281: QUESTION: math.Floor and math.Log Error?

hollowaykeanho, 2022-11-10:
Hi team! Is the following a bug?

While developing a TinyGo friendly standard library, I encountered these errors when attempting to re-use standard library's `math` package, which was marked usable and tested in the documentation (see screenshot below):

[image]

```
tinygo:ld.lld: error: undefined symbol: log
>>> referenced by log.go:81 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/log.go:81)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-7AEE18076F98FE7D131EDF8E8D934891C3E6B0C2:(hestia/hestiaSTRING.Test_S64_ParseFLOAT64$1)

tinygo:ld.lld: error: undefined symbol: floor
>>> referenced by floor.go:14 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:14)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-7AEE18076F98FE7D131EDF8E8D934891C3E6B0C2:(hestia/hestiaSTRING.Test_S64_ParseFLOAT64$1)
failed to run tool: ld.lld
```

No worries. I'm planning to adopt+modify standard library math package into my library that is friendly to TinyGo and Rust. The only disruption is scedule.

hollowaykeanho, 2022-11-10:
Issue confirmed caused by https://github.com/tinygo-org/tinygo/issues/3282 as TinyGo cannot see the assembly codes. Question answered. Further clarifications is in ticket 3282.

aykevl, 2022-11-10:
Which target is this? Because the `math` package should be supported on all platforms (except AVR but that one has bigger problems).

hollowaykeanho, 2022-11-10:
`amd64`. Command issued was `tinygo test .` inside a string processing package.

aykevl, 2022-11-10:
Can you try running `tinygo test math`? It passes for me.

hollowaykeanho, 2022-11-10:
Failed on my side:

```
u0:hestiaSTRING$ tinygo test math
tinygo:ld.lld: error: undefined symbol: log
>>> referenced by log.go:81 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/log.go:81)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Acosh)
>>> referenced by log.go:81 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/log.go:81)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Acosh)
>>> referenced by log.go:81 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/log.go:81)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Asinh)
>>> referenced 17 more times

tinygo:ld.lld: error: undefined symbol: exp
>>> referenced by exp.go:16 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/exp.go:16)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Erf)
>>> referenced by exp.go:16 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/exp.go:16)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Erf)
>>> referenced by exp.go:16 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/exp.go:16)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Exp)
>>> referenced 9 more times

tinygo:ld.lld: error: undefined symbol: exp2
>>> referenced by exp.go:142 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/exp.go:142)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Exp2)

tinygo:ld.lld: error: undefined symbol: floor
>>> referenced by floor.go:14 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:14)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Gamma)
>>> referenced by floor.go:14 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:14)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math.Lgamma)
>>> referenced by floor.go:14 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:14)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math_test.TestFloor)
>>> referenced 1 more times

tinygo:ld.lld: error: undefined symbol: ceil
>>> referenced by floor.go:43 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:43)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math_test.TestCeil)
>>> referenced by floor.go:43 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:43)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math_test.TestCeil)

tinygo:ld.lld: error: undefined symbol: trunc
>>> referenced by floor.go:61 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:61)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math_test.TestTrunc)
>>> referenced by floor.go:61 (/home/u0/Documents/zoralab/Hestia/.monteurFS/bin/golang/src/math/floor.go:61)
>>>               /home/u0/.cache/tinygo/thinlto/llvmcache-2CEFCC256783054285B8EA72C5BF6AB1E037B087:(math_test.TestTrunc)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo2492987438/main: exit status 1
FAIL
u0:hestiaSTRING$ 
```

Yeap, all above failure because tinygo can't see the `.s` assembly source codes. 

----

Version:

```
u0:hestiaSTRING$ tinygo version
tinygo version 0.26.0 linux/amd64 (using go version go1.19.1 and LLVM version 14.0.0)
```

soypat, 2022-11-12:
I'm also getting an error running `tinygo test math` on dev:
<details>

```shell
$ tinygo version
tinygo version 0.27.0-dev linux/amd64 (using go version go1.19 and LLVM version 
$ tinygo test math
ld.lld-14: error: undefined symbol: log
>>> referenced by log.go:81 (/usr/local/go/src/math/log.go:81)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Acosh)
>>> referenced by log.go:81 (/usr/local/go/src/math/log.go:81)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Acosh)
>>> referenced by log.go:81 (/usr/local/go/src/math/log.go:81)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Asinh)
>>> referenced 17 more times

ld.lld-14: error: undefined symbol: exp
>>> referenced by exp.go:16 (/usr/local/go/src/math/exp.go:16)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Erf)
>>> referenced by exp.go:16 (/usr/local/go/src/math/exp.go:16)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Erf)
>>> referenced by exp.go:16 (/usr/local/go/src/math/exp.go:16)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Exp)
>>> referenced 9 more times

ld.lld-14: error: undefined symbol: exp2
>>> referenced by exp.go:142 (/usr/local/go/src/math/exp.go:142)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Exp2)

ld.lld-14: error: undefined symbol: floor
>>> referenced by floor.go:14 (/usr/local/go/src/math/floor.go:14)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Gamma)
>>> referenced by floor.go:14 (/usr/local/go/src/math/floor.go:14)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math.Lgamma)
>>> referenced by floor.go:14 (/usr/local/go/src/math/floor.go:14)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math_test.TestFloor)
>>> referenced 1 more times

ld.lld-14: error: undefined symbol: ceil
>>> referenced by floor.go:43 (/usr/local/go/src/math/floor.go:43)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math_test.TestCeil)
>>> referenced by floor.go:43 (/usr/local/go/src/math/floor.go:43)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math_test.TestCeil)

ld.lld-14: error: undefined symbol: trunc
>>> referenced by floor.go:61 (/usr/local/go/src/math/floor.go:61)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math_test.TestTrunc)
>>> referenced by floor.go:61 (/usr/local/go/src/math/floor.go:61)
>>>               /home/pato/.cache/tinygo/thinlto/llvmcache-68652BC5248C68C90C92EDDE09430B640DBCEE46:(math_test.TestTrunc)
error: failed to link /tmp/tinygo225708749/main: exit status 1
```

</details>

aykevl, 2022-11-17:
> Yeap, all above failure because tinygo can't see the `.s` assembly source codes.

Again, _this is not the problem_. It works fine on my side.

One thing you could try is running `tinygo clean`. It might be that you have an old cache.

Other than that, you will need to provide some more steps to reproduce. How did you install TinyGo? What Linux distro are you on? Did you build from source or download the release tarball, or even installed a packaged version?

soypat, 2022-11-17:
`tinygo clean` did it for me :) 

I'll leave my installation steps for posterity:
I followed the build-drom-source steps shown on the site as of today. Since I am on linux mint I had to additionally do `sudo apt install build-essential` for a cpp compiler and stuff. I also ran `make llvm-source`. And with that I had a working tinygo installation.

aykevl, 2022-11-17:
> `tinygo clean` did it for me :)

Great! Then it's probably indeed just an old cache.
(This is still technically a bug in TinyGo as it shouldn't use a stale cache, but it's not related to assembly files).

hollowaykeanho, 2022-11-17:
Noted. Should I retitle to reflect the latest finding?

dgryski, 2024-10-21:
TinyGo cache issues have been cleaned up for a while.  Closing this unless someone can demonstrate it's still happening.

aykevl, 2024-10-22:
It's still a problem: TinyGo does use a stale cache for library builds like musl. The reason people haven't seen this for a while is probably because not a lot of files have been added to musl for a while.

One fix could be to include the list of files in the cache path as a hash.