tinygo-org/tinygo #3282: QUESTION: Has TinyGo Support Go<-->ASM codes?

hollowaykeanho, 2022-11-10:
Has TinyGo supports Go<-->ASM (`.s`) code directly?

Example:

 (`float_asm.go` <--> `float_amd64.s`)

https://cs.opensource.google/go/go/+/refs/tags/go1.19.2:src/math/floor_asm.go;bpv=0;bpt=0 <--> https://cs.opensource.google/go/go/+/refs/tags/go1.19.2:src/math/floor_amd64.s;bpv=0;bpt=0

----

I strongly believe this is the cause of this issue: https://github.com/tinygo-org/tinygo/issues/3281

Reason:

1. When I use pure-go software algorithm, everything is fine. When I port the hardware assembly optimization, the same set of errors appeared.

hollowaykeanho, 2022-11-10:
I just found https://tinygo.org/docs/concepts/compiler-internals/inline-assembly/

Just to double confirm, will this be in the correct direction? No worries, I can adjust build flags accordingly.

aykevl, 2022-11-10:
> Has TinyGo supports Go<-->ASM (`.s`) code directly?

Not yet, see https://github.com/tinygo-org/tinygo/pull/3103.

But your question is about the math package, which should work on all platforms, so I'm moving the discussion there.

closed 2022-11-10 by aykevl