tinygo-org/tinygo #3283: tinygo 0.26.0 doesn't build docs example

juliaogris, 2022-11-12:
After the upgrade from tiny go 0.25.0 to 0.26.0 and can't get the WASM example from the docs https://tinygo.org/docs/guides/webassembly/ to build anymore:

```
$ tinygo version
tinygo version 0.25.0 darwin/amd64 (using go version go1.19 and LLVM version 14.0.0)
$ tinygo build -o wasm.wasm -target wasm ./main.go # ✅

# upgrade....

$ tinygo version
tinygo version 0.26.0 darwin/amd64 (using go version go1.19.3 and LLVM version 14.0.0)
$ tinygo build -o wasm.wasm -target wasm ./main.go  # 💥
tinygo:wasm-ld: error: /var/folders/43/6hgjz_r92gv_c5mbtjx2fj5r0000gn/T/tinygo3399452899/main.o: undefined symbol: main.add
failed to run tool: wasm-ld
error: failed to link /var/folders/43/6hgjz_r92gv_c5mbtjx2fj5r0000gn/T/tinygo3399452899/main: exit status 1
```

This has also been reproduced on GitHub Actions:
successful build with 0.25.0 (go 1.19.3), failed build with 0.26.0 (go 1.19.3)
https://github.com/juliaogris/tinygo-test/actions

[image]

Upgrade commit:
https://github.com/juliaogris/tinygo-test/commit/64bea1d2413c5c28cacb73387599aace44002ae8

sago35, 2022-11-13:
@juliaogris 
I think the document needs to be updated.
Would adding `//export add` solve the problem?

see #3133

juliaogris, 2022-11-13:
Yes! that worked, thank you. In order to get the while docs sample to work again I believe another change needs to happen in JS:

```js
go.importObject.env = {
    'main.add': function(x, y) {
        return x + y
    }
    // ... other functions
}
```

needs to have `main.` removed:

```js
go.importObject.env = {
    'add': function(x, y) {
        return x + y
    }
    // ... other functions
}
```

deadprogram, 2023-08-18:
I think that was addressed already in a previous release. Please reopen if needed. Thanks!

closed 2023-08-18 by deadprogram