tinygo-org/tinygo #3284: Build wasm fail

mengdu, 2022-11-14:
os: MacOS 11.2.3, i7

Install

```sh
tar xvzf tinygo-0.26.0.darwin-amd64.tar.gz
export PATH=<extract location>/tinygo/bin:$PATH

tinygo version 
tinygo version 0.26.0 darwin/amd64 (using go version go1.18 and LLVM version 14.0.0)
```

```go
// main.go
package main

import "fmt"

// This calls a JS function from Go.
// WASMOPT=1 tinygo build -o wasm.wasm -target wasm ./main.go
func main() {
	fmt.Println("Hello")
}
```

Build

```sh
tinygo build -o wasm.wasm -target wasm ./main.go
error: could not find wasm-opt, set the WASMOPT environment variable to override
```

What is `WASMOPT`? No explanation was found in the document.

dgryski, 2022-11-15:
In order to support goroutines on wasm, tinygo uses `asyncify` which is implemented by the `wasm-opt` tool that is part of `binaryen`:  https://github.com/WebAssembly/binaryen

If the `wasm-opt` tool is not found in your path, then it uses the `WASMOPT` environment variable.  If that is not found, then it errors out.

mengdu, 2022-11-16:
Okay, thanks

closed 2022-11-16 by mengdu