tinygo-org/tinygo #3288: Allow board to be defined externally from TinyGo's main repo

waj334, 2022-11-15:
Currently, boards are defined in the TinyGo repository directly. This is good for off-the-shelf boards that can be purchased by anyone, but this model is no good for those of us who have custom boards and want to use TinyGo and I would be surprised if the original intent was to specifically add support for EVERY SINGLE BOARD IN EXISTANCE. There is some coupling within TinyGo's internals (mostly the machine package) that kept me from being able to define my board from within my project, some of which I was able to deal with:

1. The machine package assumes default pins for serial interfaces.
    - The machine package will use predefined pins for the first instance of a serial interface if no pins are specified by the respective config struct for the serial interface type.
        - Should instead the behavior be to  return an error if the config is incorrect?

**My solution**: Returning an new error (ErrInvalidConfiguration) is how I was able to overcome this.

2.  The USB implementation relies on constants defined by a board in the machine package.
    - The constants usb_STRING_PRODUCT and usb_STRING_MANUFACTURER are defined by each board in the machine package and the usb implementation will not compile if these constants are not present in the machine package for any reason.
        - Should these instead be set to some generic defaults and then able to be overridden later?
        - Should the USB implementation accept a config struct to allow a dev to specify these strings?

**My solution**: I disabled USB support in order to get the machine package to compile as USB support wasn't a priority for me (yet).

3. The default serial output cannot be changed because easily because the type that "machine.Serial" will be changes depending on build tags.
    - I honestly do not understand why there is no "Serialer" interface for serial output. Surprisingly the usb implementation will set the type it's "Serialer" interface type? Why not just have the machine package define an interface for anyone to implement as they please?

**My solution**: Just define a "Serialer" interface so that the type of the default serial interface doesn't matter so much.

With my above "solutions" I was able to get TinyGo running on my custom board without any build tag for a specific board set. I think that all boards definitions probably  should be defined within a separate package (or repo entirely like drivers) to prevent unnecessary coupling and I think that the machine package (and runtime package) should focus solely on chip support. It is my opinion that I should be able to assume the responsibility of fully initializing my chip/board if I want to and therefore however I want to (within reasonable constraints).

Other nice-to-haves:
1. Have runtime optionally NOT setup clocks for me.
    - Currently the runtime package does this, but I think clock set up should be the responsibility of the developer.
    - The machine package or board (package) can provide a default clock set up function that can be optionally used.
2. Have the compiler look in the directory of the "main" package for target json files and linker files.
    - This will benefit anyone who wants to set up memories differently based their specific use case (bootloaders, chip variants, etc...).
    
    
For additional context: My custom board is based on ATSAME51G19A which is not a supported chip in mainstream TinyGo (yet). All of this stuff were issues I ran into while resisting adding specific support for my custom board nobody else will develop for directly.

soypat, 2022-11-22:
Thank you for filing this issue. It's given me quite a bit to think about API design in the hardware world! I'd have to chew on it a bit before giving any meaningful feedback, but here are a few drive by possible solutions:

Default pins can coexist with custom board definitions if we wrap configuration functions.

```go
//go:build board
func (u UART) Configure(c UARTConfig) error {
    // Set Default Pins here
    return u.configure(c)
    }
```
```go
//go:build microcontroller
func (u UART) configure(c UARTConfig) error {
     // Return error for invalid pins
     // configure UART normally
    }
```
You could then use the `go:linkname` compiler directive/pragma to call `UART.configure` from your custom board definition with your custom default pins if you so desire. (Not sure if directive works for methods)

As for point 2., I'm not that familiar with tinygo's USB implementation.

Point 3: Yup, that's how it do be. I think the Serialer interface would belong in the tinygo/drivers package, communication interfaces are defined over there, but not in machine package.

> I think that all boards definitions probably should be defined within a separate package (or repo entirely like drivers) to prevent unnecessary coupling and I think that the machine package (and runtime package) should focus solely on chip support.

I do think this would be a valid approach, though it would require a lot of reworking of the tinygo repo. It might be for the best though as it would probably prevent these kind of usage issues in the long run. There's a lot of work to do in the embedded side of Tinygo- it has been needing some love for some time now.

As for your last two points, I got nothing, I'm kind of a noobie here.

aykevl, 2022-12-08:
Yeah I agree, and I've been thinking about how to do this for a while now.
Unfortunately, it's somewhat difficult to change now because it would break so much existing code. But I agree, we eventually have to support boards.

Duplicate of: #2607 and #2239 (and perhaps others)

davecheney, 2026-10-26:
Duplicate of #2607

closed 2026-10-26 by davecheney