tinygo-org/tinygo #3296: Unable to flash with USBtiny

mytechnotalent, 2022-11-19:
I have a USBtiny programmer and it does not work with the default
```
tinygo flash -target=arduino
error: unable to locate a serial port
```
Is there a programmer flash such as `-programmer=USBtiny` or something that will work?  (USBtiny does not work).

Thanks in advance!

aykevl, 2022-11-19:
What is this for board? If it's an Arduino Uno then flashing should work as usual (with the USB cable).

mytechnotalent, 2022-11-19:
Hi @aykevl this is a custom built Arduno where I have an ATMEGA328P on a breadboard with a crystal.  The only way to program it is to use the USBtiny.  I do try to flash with the parameters listed above however it fails.  I am able to flash a real Arduino without issue it is just an issue with USBtiny and was hoping there might be a work around.

Here is the link of the page to set up the drivers that I used. [USBtiny setup](https://learn.sparkfun.com/tutorials/tiny-avr-programmer-hookup-guide/driver-installation)

aykevl, 2022-11-19:
I see. What you could do instead is build the hex using `tinygo build -o firmware.hex` and flash it manually.
We currently don't support USBTiny itself.

mytechnotalent, 2022-11-19:
@aykevl thank you so much!  I will do just that.  I am SO excited to get onboard TinyGo!  I appreciate your fast response.  I could probably figure out how to flash with OpenOCD or something.

mytechnotalent, 2022-11-19:
The challenge is USBtiny does not provide a COM port or USB port so getting the hex on the chip will be a challenge.  Thanks so much for the help nonetheless.

closed 2022-11-19 by mytechnotalent

mytechnotalent, 2022-11-19:
@aykevl I got it to work and wanted to share with the community in case they wanted to try in addition to providing you the info if you wanted to eventually add this into a `programmer=USBtiny` someday.

```
avrdude -p m16u2 -P usb -c usbtiny -U flash:w:firmware.hex -F -v 
```