tinygo-org/tinygo #3304: Following building instructions, tinygo building gives undefined references

Ultracoolguy, 2022-11-21:
Title says it all. Afaict I've stayed as much as possible to what the BUILDING.md says, and have made no modifications to the code atm. This is on an Artix Linux installation. I'll provide any other information wanted.

```
CGO_CPPFLAGS="-I/home/me/Sources/tinygo/llvm-project/llvm/include -I/home/me/Sources/tinygo/llvm-build/include -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/me/Sources/tinygo/llvm-build/tools/clang/include -I/home/me/Sources/tinygo/llvm-project/clang/include -I/home/me/Sources/tinygo/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++14" CGO_LDFLAGS="-L/home/me/Sources/tinygo/llvm-build/lib -lclang -Wl,--start-group -lclangAnalysis -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -Wl,--end-group -lstdc++ -Wl,--start-group -llldCOFF -llldCommon -llldELF -llldMachO -llldMinGW -llldWasm -Wl,--end-group -L/home/me/Sources/tinygo/llvm-build/lib  -lLLVMWindowsManifest -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMExtensions -lLLVMLibDriver -lLLVMOption -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMOrcTargetProcess -lLLVMOrcShared -lLLVMDebugInfoPDB -lLLVMCoverage -lLLVMCoroutines -lLLVMAVRDisassembler -lLLVMAVRAsmParser -lLLVMAVRCodeGen -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyUtils -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVAsmParser -lLLVMRISCVCodeGen -lLLVMRISCVDesc -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64AsmParser -lLLVMAArch64CodeGen -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMAsmParser -lLLVMARMCodeGen -lLLVMipo -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMFrontendOpenMP -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86TargetMCA -lLLVMMCA -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMCFGuard -lLLVMGlobalISel -lLLVMX86Desc -lLLVMX86Info -lLLVMMCDisassembler -lLLVMSelectionDAG -lLLVMInstrumentation -lLLVMAsmPrinter -lLLVMDebugInfoMSF -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMDebugInfoDWARF -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMBinaryFormat -lLLVMSupport -lLLVMDemangle -lrt -ldl -lm -lxml2 -lstdc++ "  go build -buildmode exe -o build/tinygo -tags "byollvm osusergo" -ldflags="-X github.com/tinygo-org/tinygo/goenv.GitSha1=`git rev-parse --short HEAD`" .
# tinygo.org/x/go-llvm
cgo-gcc-prolog:55:11: warning: 'LLVMAddAlias' is deprecated: Use LLVMAddAlias2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:2388:1: note: 'LLVMAddAlias' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:626:11: warning: 'LLVMBuildCall' is deprecated: Use LLVMBuildCall2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:3991:1: note: 'LLVMBuildCall' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:1029:11: warning: 'LLVMBuildGEP' is deprecated: Use LLVMBuildGEP2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:3898:1: note: 'LLVMBuildGEP' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:1115:11: warning: 'LLVMBuildInBoundsGEP' is deprecated: Use LLVMBuildInBoundsGEP2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:3903:1: note: 'LLVMBuildInBoundsGEP' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:1248:11: warning: 'LLVMBuildInvoke' is deprecated: Use LLVMBuildInvoke2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:3692:1: note: 'LLVMBuildInvoke' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:1352:11: warning: 'LLVMBuildLoad' is deprecated: Use LLVMBuildLoad2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:3891:1: note: 'LLVMBuildLoad' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:1682:11: warning: 'LLVMBuildPtrDiff' is deprecated: Use LLVMBuildPtrDiff2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:4024:1: note: 'LLVMBuildPtrDiff' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:1971:11: warning: 'LLVMBuildStructGEP' is deprecated: Use LLVMBuildStructGEP2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:3908:1: note: 'LLVMBuildStructGEP' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:2657:11: warning: 'LLVMConstGEP' is deprecated: Use LLVMConstGEP2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:2156:1: note: 'LLVMConstGEP' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
cgo-gcc-prolog:2699:11: warning: 'LLVMConstInBoundsGEP' is deprecated: Use LLVMConstInBoundsGEP2 instead to support opaque pointers [-Wdeprecated-declarations]
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Core.h:2163:1: note: 'LLVMConstInBoundsGEP' has been explicitly marked deprecated here
/home/me/Sources/tinygo/llvm-project/llvm/include/llvm-c/Deprecated.h:26:23: note: expanded from macro 'LLVM_ATTRIBUTE_C_DEPRECATED'
# github.com/tinygo-org/tinygo
/usr/lib/go/pkg/tool/linux_amd64/link: running clang++ failed: exit status 1
/usr/bin/ld: /home/me/Sources/tinygo/llvm-build/lib/libclangFrontendTool.a(ExecuteCompilerInvocation.cpp.o): in function `clang::CreateFrontendAction(clang::CompilerInstance&)':
ExecuteCompilerInvocation.cpp:(.text._ZN5clang20CreateFrontendActionERNS_16CompilerInstanceE+0x23d): undefined reference to `vtable for clang::arcmt::MigrateSourceAction'
/usr/bin/ld: ExecuteCompilerInvocation.cpp:(.text._ZN5clang20CreateFrontendActionERNS_16CompilerInstanceE+0x251): undefined reference to `clang::arcmt::FileRemapper::FileRemapper()'
/usr/bin/ld: ExecuteCompilerInvocation.cpp:(.text._ZN5clang20CreateFrontendActionERNS_16CompilerInstanceE+0x6be): undefined reference to `vtable for clang::ento::AnalysisAction'
/usr/bin/ld: ExecuteCompilerInvocation.cpp:(.text._ZN5clang20CreateFrontendActionERNS_16CompilerInstanceE+0xa88): undefined reference to `clang::arcmt::CheckAction::CheckAction(std::unique_ptr<clang::FrontendAction, std::default_delete<clang::FrontendAction> >)'
[... cut ...]
/usr/bin/ld: ExecuteCompilerInvocation.cpp:(.text._ZN5clang25ExecuteCompilerInvocationEPNS_16CompilerInstanceE+0x244): undefined reference to `clang::ento::printEnabledCheckerList(llvm::raw_ostream&, clang::CompilerInstance&)'
/usr/bin/ld: ExecuteCompilerInvocation.cpp:(.text._ZN5clang25ExecuteCompilerInvocationEPNS_16CompilerInstanceE+0x253): undefined reference to `clang::ento::printAnalyzerConfigList(llvm::raw_ostream&)'
clang-14: error: linker command failed with exit code 1 (use -v to see invocation)

make: *** [Makefile:266: tinygo] Error 2
```

soypat, 2022-11-22:
I know very little about building with your own llvm, but maybe you need to add the `llvm14` build tag for the go build tool?
```
...  -tags "byollvm osusergo llvm14"   ...
```

Ultracoolguy, 2022-11-23:
Removing byollvm and adding that did made it work.

My issue is technically solved but I assume this means there is a problem with the bundled llvm being too new or smth, maybe.

aykevl, 2022-12-08:
> Removing byollvm and adding that did made it work.

Don't do that. It will cause problems (as you already found).
There are two ways to build TinyGo: either by using the LLVM branch that we use, or by using LLVM from the distribution/Homebrew. Which one do you want? You can't mix them.
In any case, please look at https://tinygo.org/docs/guides/build/ instead of BUILDING.md.