tinygo-org/tinygo #3319: tinygo with work kubernetes crds?

zirain, 2022-11-29:
I want to use TinyGo with following code:
```golang
package main

import (
	"fmt"

	workloadv1alpha1 "github.com/karmada-io/karmada/examples/customresourceinterpreter/apis/workload/v1alpha1"
	configv1alpha1 "github.com/karmada-io/karmada/pkg/apis/config/v1alpha1"
	"github.com/karmada-io/karmada/pkg/resourceinterpreter/customizedinterpreter/webhook"
	"k8s.io/apimachinery/pkg/apis/meta/v1/unstructured"
	"k8s.io/apimachinery/pkg/runtime"
)

func main() {

}

func Interpreter(req *webhook.RequestAttributes) *webhook.ResponseAttributes {
	switch req.Operation {
	case configv1alpha1.InterpreterOperationInterpretReplica:
		return responseWithExploreReplica(req)
	default:
		return nil
	}
}

func responseWithExploreReplica(req *webhook.RequestAttributes) *webhook.ResponseAttributes {
	wantedWorkload := &workloadv1alpha1.Workload{}
	if err := convertToTypedObject(req.Object, wantedWorkload); err != nil {
		return nil
	}
	return &webhook.ResponseAttributes{
		Replicas: *wantedWorkload.Spec.Replicas + 1,
	}
}

// convertToTypedObject converts an unstructured object to typed.
func convertToTypedObject(in, out interface{}) error {
	if in == nil || out == nil {
		return fmt.Errorf("convert objects should not be nil")
	}

	switch v := in.(type) {
	case *unstructured.Unstructured:
		return runtime.DefaultUnstructuredConverter.FromUnstructured(v.UnstructuredContent(), out)
	case map[string]interface{}:
		return runtime.DefaultUnstructuredConverter.FromUnstructured(v, out)
	default:
		return fmt.Errorf("convert object must be pointer of unstructured or map[string]interface{}")
	}
}
```

but got error like following:
```console
# [k8s.io/apimachinery/third_party/forked/golang/reflect](http://k8s.io/apimachinery/third_party/forked/golang/reflect)
../../../../../../pkg/mod/k8s.io/apimachinery@v0.25.4/third_party/forked/golang/reflect/deep_equal.go:126:15: v1.UnsafeAddr undefined (type "reflect".Value has no field or method UnsafeAddr)
../../../../../../pkg/mod/k8s.io/apimachinery@v0.25.4/third_party/forked/golang/reflect/deep_equal.go:127:15: v2.UnsafeAddr undefined (type "reflect".Value has no field or method UnsafeAddr)
../../../../../../pkg/mod/k8s.io/apimachinery@v0.25.4/third_party/forked/golang/reflect/deep_equal.go:314:15: v1.UnsafeAddr undefined (type "reflect".Value has no field or method UnsafeAddr)
../../../../../../pkg/mod/k8s.io/apimachinery@v0.25.4/third_party/forked/golang/reflect/deep_equal.go:315:15: v2.UnsafeAddr undefined (type "reflect".Value has no field or method UnsafeAddr)
make: *** [Makefile:7: build-wasm] Error 1
```

How can I make it work?

dgryski, 2022-12-06:
TinyGo still has issues with `tinygo` support, but in this particular case it seems like it's trying to use the vendored upstream Go `reflect` package which will definitely not work with tinygo.

zirain, 2022-12-06:
any plan to support it? it will be useful to project like https://www.kubewarden.io/

dgryski, 2022-12-06:
We can improve `reflect` support for the tinygo parts, but we won't be able build and run the upstream forked Go package. The `reflect` internals are too tied to the compiler.

zirain, 2022-12-08:
will https://github.com/tinygo-org/tinygo/pull/2640 fix `type "reflect".Value has no field or method UnsafeAddr` ?

aykevl, 2022-12-21:
> TinyGo still has issues with `reflect` support, but in this particular case it seems like it's trying to use the vendored upstream Go `reflect` package which will definitely not work with tinygo.

It's actually not that bad. It's called "forked" but in fact it's just a wrapper around the reflect package. Adding `UnsafeAddr` should be relatively easy.

> will #2640 fix `type "reflect".Value has no field or method UnsafeAddr` ?

No. It is a refactor and therefore does not add new functionality.

zirain, 2022-12-22:
yes, I take a quick look of the implement in tinygo, seems the root cause is unimplement of `Addr()` or `UnsafeAddr`
thanks for your follow up

zirain, 2023-02-20:
@aykevl #2640 is merged, can you take a look at this now?

zirain, 2023-02-23:
#3470 implement `Addr()`, any plan about `UnsafeAddr`?

dgryski, 2023-02-23:
I will work on getting these implemented, but it's just me working on it and `reflect` has a lot of moving parts.

zirain, 2023-03-14:
xref: https://github.com/tinygo-org/tinygo/pull/3554 https://github.com/tinygo-org/tinygo/pull/3560

dgryski, 2023-03-14:
Note that these particular issues fix the build issues but for full functionality all the outstanding reflect PRs should be merged.

zirain, 2023-03-15:
I test with latest commit on `dev`, still got `type "reflect".Value has no field or method UnsafeAddr`, that surprise me.

aykevl, 2023-05-03:
@zirain it's in the dev branch. Are you sure you're using the correct `tinygo` binary? (Try `tinygo clean` and then `tinygo env TINYGOROOT` to verify it points to the repository).

zirain, 2023-05-04:
get more errors about:
```
vendor/k8s.io/utils/net/port.go:124:20: undefined: net.ResolveUDPAddr
vendor/k8s.io/utils/net/port.go:128:20: undefined: net.ListenUDP
```

aykevl, 2023-05-04:
Those are different errors. We currently don't fully support the net package.

closed 2023-06-21 by zirain