tinygo-org/tinygo #3333: reflect Types (typecode) doesn't match for pointers and non-pointers

jsimnz, 2022-12-11:
I have the following example, which should (based on my understanding) print the same value. Using the `reflect.Indirect` on the given pointer type should return the same type `map[string]any`. This also doesn't work if instead of `Indirect` I use `Elem()`.

Is my understanding on the expected Types (typecode) incorrect, is there some weird behavior within the reflect package, or something else?

*note: I cast to `uintptr` just to look at a concrete value, as `RawType()` alone isn't very transparent.*

```go
package main

import (
	"reflect"
	"fmt"
)

func main() {
	m0 := make(map[string]any)
	m1 := &m0
	m0v := reflect.ValueOf(m0)
	m1v := reflect.ValueOf(m1)
	fmt.Println(uintptr(reflect.Indirect(m1v).RawType())) // prints 333
	fmt.Println(uintptr(m0v.RawType())) // print 365
}
```

aykevl, 2022-12-11:
Do not use `RawType`. It's an internal function that's only exported because it is needed in the runtime package. Therefore, I do not consider this a bug.

If you are hitting an actual bug (that doesn't involve `RawType`) then comment here and I can reopen it.

jsimnz, 2022-12-12:
I was just using `RawType` so I could cast to `uintptr`. The problem I was trying to highlight is that the Types don't line up. Even tho ugh one is the concrete type `map[string]any` and the other is the indirection of `*map[string]any`.

Here is the updated snippet
```go
func main() {
	m0 := make(map[string]any)
	m1 := &m0
	m0v := reflect.ValueOf(m0)
	m1v := reflect.ValueOf(m1)
	fmt.Println(uintptr(reflect.Indirect(m1v).RawType()))
	fmt.Println(uintptr(m1v.Elem().RawType()))
	fmt.Println(uintptr(m0v.RawType()))
	fmt.Println(reflect.Indirect(m1v).Type() == m0v.Type())
}
```

My understanding is that comparing the two `Type()` calls should be equal?

jsimnz, 2022-12-12:
> go and tinygo produce the same output:

Interesting.... running your code on `play.tinygo.org` (which Ive been using this whole time) produces different results (returns `false`)

and on my local tinygo install both produce `false`

local system:
> tinygo version
> tinygo version 0.27.0-dev-7aca8149 linux/amd64 (using go version go1.18.3 and LLVM version 15.0.0)

aykevl, 2022-12-12:
Oh, you are correct. I was using an older TinyGo version that didn't have this problem. I can reproduce it now.

jsimnz, 2022-12-12:
Thanks for looking into it.

Looking forward to when #2640 is merged 👍

dgryski, 2023-02-24:
#2640 is merged.  This will be fixed in the next release.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you!

closed 2023-06-14 by deadprogram