tinygo-org/tinygo #3340: Proposal: auto-detect target for main packages

bgould, 2022-12-16:
TinyGo already allows for providing a custom target JSON file via the -target flag.  I propose adding a feature to `tinygo build` and related tools so that if the main package contains a file called `tinygo.target.json`, by default we use that file as the input to the `-target` flag if it is not otherwise specified.

Example of how this might look in practice: https://github.com/bgould/keyboard-firmware/tree/b7ba1253bacfe848028ea58bd8c81b3a71602749/devices/macropad-rp2040

In the example, my program is always meant for a specific target, the macropad-rp2040. In this case, the `tinygo.target.json` can be stored in version control, and with the above feature it would be possible to drop into the folder and write the firmware with a simple `tinygo flash` command. In case I need to specify build tags etc for my project, this is also useful because those can go into the target file as well.  I think a lot of embedded projects are only going to have a single target file and might find this setup useful, and I presume most WASM projects would fall into that category.

This would be a useful feature even with multiple possible/suggested targets, for example:

https://github.com/bgould/keyboard-firmware/tree/b7ba1253bacfe848028ea58bd8c81b3a71602749/devices/kintqt

In the above a couple of possible target files are stored in version control; `tinygo.target.json` could be added to .gitignore, and then "selecting" a default target is as simple as making a symlink from the chosen target to `tinygo.target.json`. For this scenario it is also possible to extend `tinygo targets` command to have options like `tinygo targets select <target>` (for managing the symlink appropriately based on the host system) or `tinygo targets add <target>` (which could create a boilerplate JSON that inherits from the chosen target).

Based on feedback I can work on implementing a draft version of this feature. If you think this feature might be useful or have suggestions please let me know.  Or if you think this idea is flawed or not worth it I would appreciate that feedback as well.  Thank you.

dgryski, 2022-12-16:
We need to be watchful of any security issues this might raise, specifically around running tools or flags mentioned in the target file (target emulators, for example.).

scottfeldman, 2022-12-17:
Feature would be useful for projects with a single target, seem to me.  I like the idea of specifying the target with a file in the main dir.  

Adding build tags or other stuff to tinygo.target.json file is a little beyond what you can set with the -target flag, so that feels weird.  A file name tinygo.json might be better to specify target and other stuff?  How does -target, -tags, -etc flags given on the command line work if there is also a .json file?  Which ones wins?

Symlinks to choose between multiple target works.  Extending the tinygo target command to manage symlinks to pick a target seems like too much, when that's what -target does.

bgould, 2022-12-18:
> Adding build tags or other stuff to tinygo.target.json file is a little beyond what you can set with the -target flag, so that feels weird. A file name tinygo.json might be better to specify target [...]

With respect to this logic, I am not proposing any new functionality other than to automatically pass a particular filename as an argument for the `-target` flag; the format of that file will be the same as if that flag was specified explicitly.  Build tags, flash/run commands (as mentioned by @dgryski ), etc can already be defined/overridden in target JSON files, in my original proposal would continue to be supported.  I do agree that adding symlink management might be overkill.

That said ... @dgryski has a valid point about "automagically" referencing executables. I am still considering options but one idea I have had would be: when auto-detecting the target file, ignore any "dangerous" options in the file and emit a warning about any options that were ignored. If the user trusts the target files and wants to utilize those options, they simply would need to specify the `-target` flag explicitly, which is exactly how things work now already.

aykevl, 2023-01-10:
Yes, we need a feature like this! (In fact, I've been thinking about this with some of the design decisions).

Can we bikeshed about the name a bit though? :smile_cat: What about making it a hidden file, like `.tinygo.target.json`? (Not sure whether that's an advantage or a disadvantage here).

Regarding the multiple target features: that might be useful as well but I'd avoid this in the first pass. This is already possible using symlinks as you described if it's needed, or by specifying the JSON file directly using the `-target` flag on the command line. It's easy to add it in the future if needed, it's hard to remove a feature.

> We need to be watchful of any security issues this might raise, specifically around running tools or flags mentioned in the target file (target emulators, for example.).

Yes indeed. However I did try to keep these easy to separate. For example:

- `flash-command` is a security issue as it runs arbitrary code
- `openocd-interface` _should_ not have any security issues

I recommend having a function that whitelists properties. Something like:

```go
func sanitizeTarget(spec *TargetSpec) *TargetSpec {
    return &TargetSpec{
        Inherits: spec.Inherits,
        Triple:   spec.Triple,
        CPU:      spec.CPU,
        // etc
    }
}
```

> Adding build tags or other stuff to tinygo.target.json file is a little beyond what you can set with the -target flag, so that feels weird. A file name tinygo.json might be better to specify target [...]

I think it's fine, it works no different from a default value for `-target` (except for the whitelisting of course). For example, `-tags` on the command line just add tags to the list of build tags.

soypat, 2023-02-24:
scottfeldman mentioned they'd like to include some tinygo specific compiler flags that are required to succefully run the project such as `stack-size`. Maybe the program requires the absence of a scheduler to fit on the specific board, etc.

This would make development super nice.