tinygo-org/tinygo #3344: `FieldByName()`, `FieldByNameFunc()` in reflect.type.go

jozef-slezak, 2022-12-17:
Hello, is it ok to add FieldByName, FieldByNameFunc to reflect.type.go in tinygo?

```
func (t rawType) FieldByName(name string) (StructField, bool) {
	numField := t.NumField()
	for i := 0; i < numField; i++ {
		field := t.Field(i)
		if field.Name == name {
			return field, true
		}
	}

	return StructField{}, false
}

func (t rawType) FieldByNameFunc(match func(string) bool) (StructField, bool) {
	numField := t.NumField()
	for i := 0; i < numField; i++ {
		field := t.Field(i)
		if match(field.Name) {
			return field, true
		}
	}

	return StructField{}, false
}
```

aykevl, 2022-12-22:
> Hello, is it ok to add FieldByName, FieldByNameFunc to reflect.type.go in tinygo?

Yes, but please wait until #2640 is merged (to avoid merge conflicts). I hope it will be merged very soon.

dgryski, 2023-03-23:
My reflect work add these calls.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you!

closed 2023-06-14 by deadprogram