tinygo-org/tinygo #3351: How do I change the usb name of the device

joesonw, 2022-12-22:
When I was using Arduino IDE, I know that I can change the name by creating a new board inside `boards.txt`. 

But is there a way to change the name with tinygo? or a better and elegant solution? or Is this covered in avrdude land?

aykevl, 2023-01-11:
Right now, this is not possible. These values are hardcoded as constants (`usb_STRING_PRODUCT`, `usb_STRING_MANUFACTURER`, etc). So for a one-off project you could just modify those strings in the appropriate src/machine/board_*.go file.

This would be nice to have though, perhaps we can make it a variable so that it's possible to use the [`-X` linker flag](https://www.digitalocean.com/community/tutorials/using-ldflags-to-set-version-information-for-go-applications).

sago35, 2023-02-21:
I would like to change my VID/PID as well.
I will try a little.

aykevl, 2023-05-03:
I believe this is now fixed in https://github.com/tinygo-org/tinygo/pull/3583.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you!

closed 2023-06-14 by deadprogram