tinygo-org/tinygo #3355: Go routine not workin on a pico

BerndCzech, 2022-12-26:
Sorry for the bolt question but **how could I debug** why go routines don´t work on my **raspberry pi pico**?

image code like

```go

func main(){
  go func(){ 
    blink_led()
  }

  readTemp()
}
```

In fact my led stays off and the temperature display shows 0.
It seems like the whole code is skipped.

aykevl, 2022-12-26:
Can you also share the code for `readTemp`?

BerndCzech, 2022-12-26:
Glad you are interested

I just use a buzzer instead of displaying a temperature to reduce the code. But the concurrency is not working here.

```go

package main

import (
	"machine"
	"tinygo.org/x/drivers/buzzer"
)

func main(){
	btn := machine.GP10
	btn.Configure(machine.PinConfig{Mode: machine.PinInput})

	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})

	// works
	// led.High()

	// does not work
	go func() {
		led.High()
	}()

	playC := func() {
		bzrPin := machine.GP20
		bzrPin.Configure(machine.PinConfig{Mode: machine.PinOutput})
		bzr := buzzer.New(bzrPin)

		bzr.Tone(buzzer.C4, buzzer.Half)
	}

	for {
		if btn.Get() {
			playC()
		}
	}
}
```

The observed behavior is: 

* LED is off  a the beginning
* Once I touch the buzzer i hear the C playing
* The led is switched on and stays on

I would expect that the LED is not off at the beginning.

I could extend the example with some concurrency - so please ask me.

bgould, 2022-12-26:
does it work if you update this
```
	// does not work
	go func() {
		led.High()
	}()
```

to be this instead?

```
	// does not work
	go func() {
		for {
			led.High()
			time.Sleep(500 * time.Millisecond)
			led.Low()
			time.Sleep(500 * time.Millisecond)
		}
	}()
```

Please remember that when you start a goroutine, once the function exits the goroutine also exits.

BerndCzech, 2022-12-26:
The behavior became now very odd:

* LED is off at the beginning
* Once I hit the button i hear the C playing
* The first time the button is pushed the led switches on and off for 1 cycle
* Any subsequent button push the LED remains off

https://user-images.githubusercontent.com/69037455/209586739-0a2b2772-a89d-4f07-ab97-d004d2c54508.mp4

Comparing both code snippets I would conclude it seems like the **go routine is run once** and never again.

BerndCzech, 2022-12-26:
I pursued this understanding and replaced

```go
  for {
    if btn.Get() {
      playC()
    }
  }
```
by 
```go
  for {
    if btn.Get() {
        playC()
      } else {
        time.Sleep(time.Millisecond)
  }
}
```
and now it works without any errors. Could it be that the infinite loop occupies the whole cpu and the go routine is not handled anymore?

I would work around the issue by with some kind of useless sleep command at the end of my programs. But is that behavior expected?

bgould, 2022-12-27:
> 
> and now it works without any errors. Could it be that the infinite loop occupies the whole cpu and the go routine is not handled anymore?
> 
> I would work around the issue with some kind of useless sleep command at the end of my programs. But is that behavior expected?

yes this is expected ... if the main loop does not yield at all with blocking operations (like the time.Sleep invocation) then control will not be passed to other goroutines.  Alternative to time.Sleep you could probably use `runtime.Gosched` (https://pkg.go.dev/runtime#Gosched) to make it work.

BerndCzech, 2022-12-27:
@bgould ty - I will close the issue now. Could you amplify what implications of your quote below are? I don´t see them myself.
 
> Please remember that when you start a goroutine, once the function exits the goroutine also exits.

closed 2022-12-27 by BerndCzech

bgould, 2022-12-27:
Please disregard ... I misinterpreted the intention of your code I believe when I made that comment. Thank you!