tinygo-org/tinygo #3362: [proposal] Support for machine/board Bundles

ya-mouse, 2023-01-02:
Within a time we will have a significant number of supported machines/boards. This will bloat the base tinygo srcs.

**Proposal**

This proposal is about to split the machines-related code in "bundles". The existing code allows to merge regular Go sources and TinyGo (btw, src/os/exec/exec.go never being used with current implementation because os/exec is overriden by Go sources). The approach is to traverse bundles `src` directories and merge the contents with the TinyGo sources in the cached root. Same should be done for `lib` and `bin` if present. The per bundle `targets` should be merged separately. 

**Dependencies**
In order to support cross-bundle dependencies, it's possible to wrap the `go install` command to be called within bundle directory (set the `GOPATH` variable). The existing PoC should be re-worked to follow pkg-structure of Go packages.

**Caching**
At the first implementation the `targets` being traversed each invocation and only missing symlinks are re-created.
The bundle modifications (install, update, remove) should invalidate the existing cache.

The PoC that splits up the `nrfx` package: https://github.com/tinygo-org/tinygo/pull/3361

aykevl, 2023-01-03:
This is a pretty drastic change. What's the advantage of this new system? Is there an advantage other than reducing the number of files in the machine package?

ya-mouse, 2023-01-03:
I will split the change. The main part is for actual bundles support and the rest is files bouncing around.
The advantage to move out machine/boards/targets out of main src tree.

aykevl, 2023-01-11:
> The advantage to move out machine/boards/targets out of main src tree.

Ah I see.  In that case, I think the easier option would be to move the machine package as a whole out of the tinygo tree. This is something I've wanted to do for a while but it is rather difficult to do and would probably break backwards compatibility with existing code.

ya-mouse, 2023-01-12:
The proposed change and working proof of concept do seamless merge of files for machine package and runtime as well. The bunch of files for machines are also there. With no backward compatibility breakage.

aykevl, 2023-01-12:
> The proposed change and working proof of concept do seamless merge of files for machine package and runtime as well.

I would very much prefer to _not_ do that. We already do some merging of the Go standard library. We already have two variants (one with the syscall package and one without), and these variants are already troublesome for things like IDE support. Please don't introduce more variants for chip families.

ya-mouse, 2023-01-13:
So, the only way is to move into separate packages out of the main tree? Then, the changes will not be back compatible. And things like uart from existing machine/runtime will require some extra init function to pass the machine's uart instance and so on.

digitalentity, 2026-11-09:
This now became a blocker for adopting TinyGo... 🫤

digitalentity, 2026-11-09:
@deadprogram I plan to tackle this and present a plan for review in the next few days.

digitalentity, 2026-11-10:
Here's what I came up with (drawing inspiration from [TamaGo](https://github.com/usbarmory/tamago)). The proposal is largely compatibie with the current target declarations and allows gradual migration.

### 1. Two-Stage Boot Hooks in `runtime.run()`

TinyGo splits hardware initialization into two stages in `runtime.run()` for all schedulers. This design does not modify architecture entry assembly.

- **`hwInit0()`** runs first in `run()`, before `initHeap()`, on core 0 and the main stack without heap allocation. It calls `socInit0()` (peripheral resets, crystal, PLLs, clocks, and timer) and then `boardInit0()` (PMIC rails, external PSRAM, and early GPIO). No heep allocation is allowed here (will result in a crash as heap doesn't exist yet).
- **`hwInit1()`** runs after `initHeap()` in the main execution context, right before `initAll()`. It calls `socInit1()` and then `boardInit1()` (board peripheral structs, IRQ handlers, `Configure` calls, and `machine.Serial`).

On targets with the `socinit` build tag, `runtime.socInit0`, `runtime.socInit1`, `runtime.boardInit0`, and `runtime.boardInit1` are `//go:weak` empty functions. `package machine` overrides `socInit0` and `socInit1` with `//go:linkname`. A board package overrides `boardInit0` and `boardInit1` with `//go:linkname`. On targets without `socinit`, `hwInit0()` and `hwInit1()` are empty functions that the compiler removes at `-opt=1` and above.

### 2. Custom Board Mode in `package machine`

The chip targets (like `rp2040` and `rp2350`) are `inheritable-only`. Therefore, each board (custom or standard) provides a small `target.json` file that inherits the chip target.

The following rules apply:

- `package machine` omits the `UART0`, `UART1`, `I2C0`, `I2C1`, `SPI0`, `SPI1` and other global variables for instantiated peripherals.
- The board package creates and configures only the peripherals that exist on the PCB.
- For `"serial": "uart"`, `package machine` declares `var Serial *UART` without an initializer, `InitSerial()` does nothing, and `boardInit1` is expected to set `machine.Serial`.
- The board package provides USB descriptor values through `machine.boardUSBIdentity` with `//go:linkname`. This function is required when the program uses USB descriptors, and optional otherwise.
- The target loader injects the external `linkerscript` if provided.

### 3. Compiler, Linker, and Interpreter Checks

- **`//go:weak` pragma.** TinyGo allows a new `//go:weak` only in `runtime` and `machine`. The builder links weak and strong definitions in any package order and checks function signatures.

## Boot Sequence Comparison (on RP2 targets)

From `Reset_Handler` to `callMain`, SoC and board initialization move before `initAll()`. Inside `initAll()`, `runtime.init()` now calls only `machine.InitSerial()`.

```mermaid
flowchart TB
    subgraph PROP["Proposed (rp2)"]
        direction TB
        PR["Reset_Handler"] --> PM["main"]
        PM --> PP["preinit (.bss, .data)"]
        PP --> PPR["prerun"]
        PPR --> PRUN["run"]
        PRUN --> PSOC0
        subgraph PHW0["hwInit0 (pre-heap)"]
            PSOC0["socInit0 (resets, clocks, PLLs, timer)"] --> PBOARD0["boardInit0 (PMIC, PSRAM, early GPIO)"]
        end
        PBOARD0 --> PIR["initRand"]
        PIR --> PIH["initHeap"]
        PIH --> PSOC1
        subgraph PHW1["hwInit1 (post-heap, main context)"]
            PSOC1["socInit1 (weak no-op on rp2)"] --> PBOARD1["boardInit1 (peripherals, IRQs, machine.Serial)"]
        end
        PBOARD1 --> PPKG1
        subgraph PIA["initAll"]
            PPKG1["package init (before runtime)"] --> PIS
            subgraph PRI["runtime.init"]
                PIS["machine.InitSerial"]
            end
            PIS --> PPKG2["package init (after runtime)"]
        end
        PPKG2 --> PCM["callMain"]
    end

    subgraph CUR["Current (rp2)"]
        direction TB
        CR["Reset_Handler"] --> CM["main"]
        CM --> CP["preinit (.bss, .data)"]
        CP --> CPR["prerun"]
        CPR --> CRUN["run"]
        CRUN --> CIR["initRand"]
        CIR --> CIH["initHeap"]
        CIH --> CPKG1
        subgraph CIA["initAll"]
[... 52 lines cut ...]
func boardLateInit() {
	auxUART.Bus = rp.UART0
	auxUART.Buffer = &auxBuf
	auxUART.Interrupt = interrupt.New(rp.IRQ_UART0_IRQ, auxUART.HandleInterrupt)
	auxUART.Configure(machine.UARTConfig{BaudRate: 115200, TX: AuxTX, RX: AuxRX})
	AuxUART = &auxUART

	StatusLED.Configure(machine.PinConfig{Mode: machine.PinOutput})
}

//go:linkname boardUSBIdentity machine.boardUSBIdentity
func boardUSBIdentity() (vid, pid uint16, manufacturer, product string) {
	return 0x1209, 0x0001, "Example Corp", "Custom Board"
}
```

Run this command from the user module root to build the firmware. You do not need to modify files in `$TINYGOROOT`.

```bash
tinygo build -target=./boards/custom/target.json ./cmd/firmware
```

## Compatibility

Existing targets and boards keep their current source API and runtime behavior. On rp2 targets, clocks start earlier in `hwInit0()`, so all package `init()` functions see the final clock frequencies. Targets without `socinit` have zero code generation changes at `-opt=1` and above.

digitalentity, 2026-11-15:
@deadprogram @aykevl  wdyt? this is a rather big refactoring of the compiler itself, so I'd appreciate a second opinion before I implement anything.