tinygo-org/tinygo #3369: unexpected token ILLEGAL when compiling <sys/socket.h> with wasi target

charconstpointer, 2023-01-07:
I was playing around with https://github.com/alphahorizonio/unisockets and I've noticed some issues when compiling its code for `wasi`, narrowed it down to the snippet below and it seems like it's the issue with `<sys/socket.h>` and underlying wasi implementation
```go
package main

/*
#include <sys/socket.h>
*/
import "C"

const (
	SHUT_RDWR = int32(C.SHUT_RDWR)
)

func main() {
}
```

Compiled with `tinygo build -target wasi main.go`
Output `tinygo/lib/wasi-libc/sysroot/include/__header_sys_socket.h:12:35: unexpected token ILLEGAL, expected end of expression`

---
Env
- Tinygo version: 0.26.0
- go version go1.19.4 linux/amd64

aykevl, 2023-01-11:
I've implemented support for `|` in https://github.com/tinygo-org/tinygo/pull/3379. It doesn't fully fix this bug though, the new error is the following:

```
$ tinygo run -target=wasi ./tmp/issue3369.go
# command-line-arguments
lib/wasi-libc/sysroot/include/__header_sys_socket.h:12:20: undeclared name: C.SHUT_RD
lib/wasi-libc/sysroot/include/__header_sys_socket.h:12:30: undeclared name: C.SHUT_WR
```

Not sure what's going on there.

davecheney, 2026-10-27:
Further investigation is blocked by #6042 for now. The reproducer still fails on current dev, and we’ve isolated a CGo macro parsing failure involving the C casts in the WASI shutdown constants. We can return to this once that is addressed.