tinygo-org/tinygo #3372: reflect.SliceHeader expects incorrect field types

bjatkin, 2023-01-07:
The following code does not compile correctly when using the tinygo compiler
```go
package main

import (
	"fmt"
	"reflect"
	"unsafe"
)

func main() {
	a := []int{0, 1, 2, 3, 4}
	b := reflect.SliceHeader{
		Data: uintptr(unsafe.Pointer(&a[0])),

		Len: len(a),
		Cap: len(a),
	}

	fmt.Println(b)
}
```

when running `tinygo run .` I get the following output
```sh
main.go:14:8: cannot use len(a) (value of type int) as uintptr value in struct literal
main.go:15:8: cannot use len(a) (value of type int) as uintptr value in struct literal
```

surprisingly however, when I change the types of the `Len` and `Cap` fields to `uintptr`s, the code complies successfully and runs correctly.

```go
func main() {
	a := []int{0, 1, 2, 3, 4}
	b := reflect.SliceHeader{
		Data: uintptr(unsafe.Pointer(&a[0])),

		Len: uintptr(len(a)),
		Cap: uintptr(len(a)),
	}

	fmt.Println(b)
}
```

This new code however, will not compile when using the `go` compiler as the SliceHeader type expects `int`s for it's length and capacity.

running`go run .` with this version of the code results in the following error
```sh
./main.go:14:8: cannot use uintptr(len(a)) (value of type uintptr) as type int in struct literal
./main.go:15:8: cannot use uintptr(len(a)) (value of type uintptr) as type int in struct literal
```

I believe the behavior of the `go` compiler is correct here and the `tinygo` compiler should be updated to expect `int` values for the `Len` and `Cap` fields in the `reflect.SliceHeader` type.

I'm using tinygo version 0.26.0 
```sh
tinygo version 0.26.0 linux/amd64 (using go version go1.19 and LLVM version 14.0.0)
```

dgryski, 2023-01-07:
Yes. Please see https://github.com/tinygo-org/tinygo/issues/1284 .

bjatkin, 2023-01-11:
I appreciate the context. I will port my code to instead use the new unsafe.Slice() function introduced in 1.17

closed 2023-01-11 by bjatkin