tinygo-org/tinygo #3375: Add (or make visible in docs) build tags for conditional compilation for target

andrewfstratton, 2023-01-09:
When a module needs to be different for different devices - e.g. badger2040 and pybadge (where the displays have different devices, etc.) it would be great to allow for something like Go's ability to have one of:

  1. filename identification for a target to allow only the relevant module to be built for a target device.  e.g. something like the go os/architecture module name - but this would probably need to be - package_os_arch_target.go.
  2. build tags in the module - though deciding on which style to use might be an issue and also some technical debt (since I think tinygo is not yet up to go:build version yet)

  I imagine that filename based is quicker, easier and incurs less technical debt (at present) and should be sufficient for now.

aykevl, 2023-01-09:
For this we use build tags. You can see the list of build tags using the `tinygo info` subcommand. For example:

```
~$ tinygo info badger2040
LLVM triple:       thumbv6m-unknown-unknown-eabi
GOOS:              linux
GOARCH:            arm
build tags:        cortexm baremetal linux arm rp2040 rp badger2040 tinygo math_big_pure_go gc.conservative scheduler.tasks serial.usb
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     /home/ayke/.cache/tinygo/goroot-3a440b66d587e4e8eab2a8bcafe00a636ab2691224cdd9e8c1274c2ea5dc6fe9
```

> build tags in the module - though deciding on which style to use might be an issue and also some technical debt (since I think tinygo is not yet up to go:build version yet)

TinyGo is Go, so this works just like you'd expect from regular Go. `//go:build` and `// +build` both work (although I'd recommend `//go:build` for new code). You can't exclude files using filename suffixes: that's not a supported feature in Go except for a number of special-cased build tags (like `arm`, `linux`, etc). There is not much TinyGo can do about this.

andrewfstratton, 2023-01-09:
Excellent :)  Also nice to know //go:build can be used as well.

Just to check - for filename suffix - is it just of the form '<package_name>_<target>.go' , e.g. 'mydisplay_badger2040.go'?

Please could this be put in the docs somewhere - maybe in the FAQ?

davecheney, 2026-10-27:
Thanks for confirming that `//go:build badger2040` works. Closing as answered. Board selection uses build tags, not board-name filename suffixes; `tinygo info <target>` lists the available tags.

closed 2026-10-27 by davecheney