tinygo-org/tinygo #3390: DIfferent behavior in Go 1.19/TinyGo when casting float with negative value to uint

deadprogram, 2023-01-16:
The following program produces different output in Go 1.19 vs. TinyGo:

```go
package main

func main() {
	var vf1 float64 = -5.865784
	var vf2 float64 = 5.865784
	vu1 := uint32(vf1 * 10000)
	vu2 := uint32(vf2 * 10000)
	println(vu1, vu2)
}
```

Running this using Go 1.19 results in:

```
4294908639 58657
```

Running it using TinyGo results in:

```
0 58657
```

aykevl, 2023-01-16:
Interesting. In this case, I think the behavior is actually correct in both cases. Namely, [the spec says the following](https://go.dev/ref/spec#Conversions):

>  In all non-constant conversions involving floating-point or complex values, if the result type cannot represent the value the conversion succeeds but the result value is implementation-dependent.

As the number -58657.84 cannot be represented as a uint32 (a negative value can never fit in an unsigned integer), the implementation is free to pick a number. Usually this is whatever the underlying architecture uses as a default value for out-of-range values but I believe we have some custom code for this in TinyGo because the native LLVM instruction results in undefined behavior on out-of-range values.
See: https://github.com/tinygo-org/tinygo/issues/276

dgryski, 2023-01-16:
So converting to a signed int type first would be "safer" in this case rather than going directly from `float64` to `uint32`?

aykevl, 2023-01-16:
Depends on what you want. If the value could be negative, converting to a `int32` (or `int64`) first at least gives you a consistent result. But converting a negative float value to an unsigned integer will always be a "weird" value.

Here is a slightly more simplified version of the reproducer:

```go
package main

func main() {
        var vf1 float64 = -5
        var vf2 float64 = 5
        println(uint32(vf1), uint32(vf2))
}
```

hollowaykeanho, 2023-01-20:
quick query: isn't this not a bug and expected or am I not getting the problem correctly?

Why I'm asking: there was a guy aggressively attacking the same query in [the other forum](https://forum.golangbridge.org/t/inconsistent-behaviour-when-converting-float-to-uint32-on-arm/30247) yesterday and I'm currently rewriting the standard libraries for TinyGo first (at the moment dealing with math packages, just finished `float->string` conversion from scratch).

I'm interested in your inputs as I can't observe the problem. If it's a legit technical bug, I wish to capture the issue in the rewritten library.

My understanding is as follows:

----

Inside mathematical domain

1. `uint` type only makes mathematical sense for positive precise values (unsigned) mathematical representations.
2. `float` is signed approximation value.
3. So converting `float` to `int` makes mathematical sense and `float` to `uint` does not when the `float` value is outside of acceptable boundary.
4. By such practices, the result would be chaotic (can be anything, `0`, `0xDEAD`, `MIN/MAX`).

----

Ouside mathematic domain

1. I'm not saying you can't do `float --> uint` for negative value but is definitely outside mathematic domain. E.g. dissecting the `float` IEEE754 bits for analysis (I did that when I was re-writing `float->string` functions).
2. `uint` can be used for bitwise operations (thus accepting negative value). When used in this context, the mathematical  and numerical represention won't be making sense.

----

Am I missing something here?

deadprogram, 2023-03-31:
Closing since this is undefined behavior. Thanks everyone!

closed 2023-03-31 by deadprogram