tinygo-org/tinygo #3394: Serial print stops working after SPI configure on pico target using a picoprobe

incredible-io, 2023-01-17:
Hello.

I have a ST7735 display connected to SPI1 at a pico and this pico connected with SWD to another pico to be able to debug my code. The following minimal sample code reproduce this issue:
```golang
package main

import (
	"machine"
	"time"
)

func main() {
	time.Sleep(time.Second * 5)
	println("Hello, TinyGo!") // This print works, 
                                  // but the last character (\n) seems not to be printed, if
                                  // directly followed by the next line without time.Sleep()

	if err := machine.SPI1.Configure(machine.SPIConfig{
		SCK: machine.SPI1_SCK_PIN,
		SDO: machine.SPI1_SDO_PIN,
	}); err != nil {
		panic(err)
	}

	for {
		print(".") // These prints are not shown in my serial monitor, 
                           // but the loop is running
		time.Sleep(time.Second)
	}
}
```
I'm using this command to flash the picoprobe:
```
$> tinygo flash -target pico -programmer picoprobe -serial uart .
```
If I comment out the block with `machine.SPI.Configure()`, all prints from the for-loop are shown in my serial monitor. So in my real code I cannot use print for logging status messages, and so on, if I use my display at the same time. If I use only one pico without the debug-probe-setup from the handbook, the sample code runs fine, also with configured SPI.

So I have no idea what I'm missing, or maybe what is wrong with the picoprobe setup?

The wiring is like in [Appendix A](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) plus the wiring from the [display](https://www.waveshare.com/wiki/Pico-LCD-1.8).

I'm using TinyGo version:
```
$> tinygo version
tinygo version 0.26.0 linux/amd64 (using go version go1.19.5 and LLVM version 14.0.0)

$> tinygo info pico
LLVM triple:       thumbv6m-unknown-unknown-eabi
GOOS:              linux
GOARCH:            arm
build tags:        cortexm baremetal linux arm rp2040 rp pico tinygo math_big_pure_go gc.conservative scheduler.tasks serial.usb // <- this is corrected within the flash command, right?
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     /home/.../.cache/tinygo/goroot-...
```

Regards,
Cid

soypat, 2023-02-12:
Hey there! It seems when you do this the SPI's `SDI` pin is configured as an input which unsets your `GPIO0` Tx pin as output, breaking your UART0 setup. To fix this add and SDI field to the SPIConfig struct with your SDI pin.
```go
// ...
SCK: machine.SPI1_SCK_PIN,
SDO: machine.SPI1_SDO_PIN,
SDI: machine.SPI1_SDI_PIN,
// ...
```

incredible-io, 2023-02-15:
Thank you, that works.

closed 2023-02-15 by incredible-io