tinygo-org/tinygo #3397: Should the main package be in one module?

vlkv, 2023-01-18:
Hello, 
Excuse me, for my maybe noob question. But when I create this code:

file: ./hello.go
```
package main

import "fmt"

func main() {
	fmt.Printf("Hello, %s", GetFoobar())
}
```

file: ./foobar.go
```
package main

func GetFoobar() string {
	return "foobar"
}
```
And try to build it with command:
```
$ tinygo build -o hello.wasm -target wasi ./hello.go
```
I get an error:
```
# command-line-arguments
hello.go:8:26: undeclared name: GetFoobar
```

Is there a requirement that all code of main package should be in one single *.go module? Or am I missing something... It seems that other packages could be split into several modules with no problems.

Thank you.

aykevl, 2023-01-18:
Go compiles code in packages, where one package is all the Go files in one directory. (A Go module is something else: a collection of packages, mostly related to versioning).

I think what you wanted to do is the following, compiling the specified directory:

```
tinygo build -o hello.wasm -target wasi .
```

You can even leave out the path to the current directory:

```
tinygo build -o hello.wasm -target wasi
```

If you specify a Go file directly, it will consider that file the whole module. It is only really used for small tests, in practice almost everybody just refers to a directory or import path, not to a file.

vlkv, 2023-01-18:
Thank you very much for the explanation!

closed 2023-01-18 by vlkv