tinygo-org/tinygo #3408: Go 1.20: archive/zip fails due to unimplemented godebug.setUpdate

QuLogic, 2023-01-23:
Running with #3391 properly this time, it appears there are only test failures in `archive/zip`:
```
--- FAIL: TestCVE202127919 (0.00s)
    Error reading the archive: <nil>
    FailNow is incomplete, requires runtime.Goexit()
--- FAIL: TestCVE202141772 (0.00s)
    Error reading the archive: <nil>
    FailNow is incomplete, requires runtime.Goexit()
--- FAIL: TestInsecurePaths (0.00s)
    NewReader for archive with file "../foo": got err <nil>, want ErrInsecurePath
    NewReader for archive with file "/foo": got err <nil>, want ErrInsecurePath
    NewReader for archive with file "a/b/../../../c": got err <nil>, want ErrInsecurePath
    NewReader for archive with file "a\\b": got err <nil>, want ErrInsecurePath
FAIL
FAIL	archive/zip	62.713s
```

This is because these three tests call `t.Setenv("GODEBUG", "zipinsecurepath=0")`, which requires refreshing the `GODEBUG` settings so that `archive/zip` will return an error. But since #3391 only added a stub for `setUpdate`, the refresh doesn't happen, no error is returned, and those tests fail.

aykevl, 2023-01-25:
Thank you for the investigation! Yes, I encountered this error while working on #3391 but didn't investigate it yet (#3391 was focused on fixing the most apparent Go 1.20 breakage, not to add full support for everything).

Feel free to submit a PR with a fix, otherwise I'll probably get around to it around the time Go 1.20 is released.

aykevl, 2023-02-02:
See: https://github.com/tinygo-org/tinygo/pull/3426

deadprogram, 2023-02-12:
0.27.0 has now been released, so closing.

closed 2023-02-12 by deadprogram