tinygo-org/tinygo #3428: USB Serial print stops working when using USB HID (RP2040)

neildavis, 2023-02-03:
Possibly related to #3394 ?

If I use a USB HID device (e.g. `machine/usb/hid/mouse`), then shortly after sending the first 'reports' (e.g. moving the mouse) output via `println()` stops being sent & received over USB serial.

Here's an example program to demonstrate:

```go
package main

import (
	"machine/usb/hid/mouse"
	"time"
)

func main() {
	moveMouse := false
	mouse := mouse.New()
	for i := 0; i < 300; i++ {
		time.Sleep(time.Millisecond * 200)
		println("Counting ...", i)
		if i > 50 && moveMouse {
			moveX := 10
			if (i % 2) > 0 {
				moveX = -10
			}
			mouse.Move(moveX, 0)
		}
	}
}
```

Once the loop counter var `i` exceeds 50 (10 secs) it starts moving the mouse and soon after the `println("Counting ...", 1)` will stop working. The mouse does continue to move the cursor on my PC though so it hasn't frozen.

if you change `moveMouse` to `false` the serial output will continue to completion.

sago35, 2023-02-03:
@neildavis 
The following fixes may eliminate the problem.
https://github.com/tinygo-org/tinygo/pull/3343

sago35, 2023-02-03:
I was able to reproduce the problem in Tinygo-0.26.
However, this did not occur in the latest dev branch.
Perhaps the problem has been fixed.

neildavis, 2023-02-04:
@sago35 Thanks, I re-installed from the [latest 0.27 build](https://github.com/tinygo-org/tinygo/actions/runs/4085636487) and can confirm this is no longer an issue.

closed 2023-02-04 by neildavis