tinygo-org/tinygo #3429: testing u-root backoff command results in nil pointer dereference

rminnich, 2023-02-03:
using github.com/u-root/u-root, v0.11.0
GOOS is linux
GOARCH is amd64
tinygo version 0.26.0 linux/amd64 (using go version go1.18.1 and LLVM version 14.0.0)

cd cmds/core/backoff

rminnich@a300:~/go/src/github.com/u-root/u-root/cmds/core/backoff$ go test -v .
=== RUN   TestRunIt
=== RUN   TestRunIt/_date
Fri Feb  3 08:47:12 PM UTC 2023
=== RUN   TestRunIt/noCmd
=== RUN   TestRunIt/echo
hi
=== RUN   TestRunIt/echo_missing_unit
--- PASS: TestRunIt (0.00s)
    --- PASS: TestRunIt/_date (0.00s)
    --- PASS: TestRunIt/noCmd (0.00s)
    --- PASS: TestRunIt/echo (0.00s)
    --- PASS: TestRunIt/echo_missing_unit (0.00s)
=== RUN   TestOK
--- PASS: TestOK (2.98s)
PASS
ok  	github.com/u-root/u-root/cmds/core/backoff	2.992s

rminnich@a300:~/go/src/github.com/u-root/u-root/cmds/core/backoff$ tinygo test -v .
panic: runtime error: nil pointer dereference
FAIL	github.com/u-root/u-root/cmds/core/backoff	3.050s
FAIL

aykevl, 2023-02-03:
Have you tried getting a backtrace using GDB? You can build a binary using `tinygo test -c` and then run it using `gdb`. When it crashes, do `backtrace` (or `bt`) to see where it crashed.

dgryski, 2023-02-03:
On my Mac with the latest dev branch I get 
```
~/go/src/github.com/u-root/u-root/cmds/core/backoff $ tinygo test
2023/02/03 14:51:01 error building go package in "/Users/dgryski/go/src/github.com/u-root/u-root/cmds/core/backoff": , fork/exec /usr/local/go/bin/go: operation not implemented
FAIL	github.com/u-root/u-root/cmds/core/backoff	0.180s
```

My guess is something either in tinygo or u-root is/was failing to check an error return.  

TinyGo does not yet support `os/exec`.