tinygo-org/tinygo #3435: When is the next tinygo release?

jmoney, 2023-02-09:
I noticed it hasn't been released since october of 2022.  Curious if there is any plan to release it soon

dgryski, 2023-02-09:
We're working on it. :) Stay tuned.

ulisesflynn, 2023-02-09:
Along these lines, will this release include an update to allow the use of Go 1.20?
`error: requires go version 1.18 through 1.19, got go1.20`

Thank you for all the great work on tinygo!

jmoney, 2023-02-09:
Looking forward to it.  i really just want https://github.com/tinygo-org/tinygo/commit/62594004c60cb3d8fab328aab71e5e0ea7e58119 in a release so i can test it out.  I have a bit of code that relies on `xml.Escape` from `encoding/xml` which currently fails in tinygo.

Mossaka, 2023-02-09:
Hey! If you just want to test it out locally, feel free to grab the TinyGo 0.27 artifacts from the CI

Note: the link below is the link to Linux TinyGo SDKs. You can find artifacts for your OS in other workflows.
https://github.com/tinygo-org/tinygo/actions/runs/4085636487#artifacts

jmoney, 2023-02-10:
I will definitely test it out and report any issues this evening.

jmoney, 2023-02-10:
This release works perfectly for my use cases!  Anyway you can get the mac binaries out?  Right now I'm doing some nonsense docker building just to generate the wasm i need.

Mossaka, 2023-02-10:
Here is a macos binary for amd64 https://github.com/tinygo-org/tinygo/actions/runs/4085636484#artifacts

jmoney, 2023-02-10:
sorry again, but the arm64 ones perhaps :D  Have an M1 unfortunately.   The others will work for our CI pipeline but locally i'll need the mac arm64 ones.

aykevl, 2023-02-11:
> sorry again, but the arm64 ones perhaps :D Have an M1 unfortunately. The others will work for our CI pipeline but locally i'll need the mac arm64 ones.

v0.27.0 (the coming release) won't have darwin/arm64 support unfortunately because GitHub Actions doesn't support this OS/arch. You have two options: build a release yourself or use Rosetta 2.

jmoney, 2023-02-11:
Interesting, so when I do `brew install tinygo` that much be installing darwin/amd64 and then using rosetta?  In which case, i can make this work with darwin/amd64.  Thanks for the note.

jmoney, 2023-02-12:
Ok, i have everything tested and working locally and in CI.  Now, when do you think that 0.27 will officially be released?  I'd like to remove some of the "hacks" i have in place in CI to get the built versions that have so generously been generated and looking to see when i'd be able to do so.

dgryski, 2023-02-12:
https://github.com/tinygo-org/tinygo/releases/tag/v0.27.0

closed 2023-02-12 by dgryski