tinygo-org/tinygo #3438: Cross compiling on ppc64le architecture.[Question]

Sunidhi-Gaonkar1, 2023-02-10:
Hi Team, I am attempting to cross compile Tinygo to build binaries for Power Linux (ppc64le) platform. Tinygo is used to run unit and integration tests by aquasecurity/trivy (https://github.com/aquasecurity/trivy). So far I have followed the steps mentioned below:
1.Installed the dependencies mentioned here: https://github.com/tinygo-org/tinygo/blob/release/BUILDING.md
2.Installed qemu and powerpc64le-linux-gnu.
3.Downloaded llvm source.

The command `make llvm-build` fails with following error:

```
[9/3024] Building CXX object lib/Support/CMakeFiles/LLVMSupport.dir/APFloat.cpp.o
FAILED: lib/Support/CMakeFiles/LLVMSupport.dir/APFloat.cpp.o
/usr/bin/clang++ -DGTEST_HAS_RTTI=0 -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/user5/tinygo/llvm-build/lib/Support -I/home/user5/tinygo/llvm-project/llvm/lib/Support -I/home/user5/tinygo/llvm-build/include -I/home/user5/tinygo/llvm-project/llvm/include -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -ffunction-sections -fdata-sections -Werror=global-constructors -O3 -DNDEBUG -std=c++14  -fno-exceptions -fno-rtti -MD -MT lib/Support/CMakeFiles/LLVMSupport.dir/APFloat.cpp.o -MF lib/Support/CMakeFiles/LLVMSupport.dir/APFloat.cpp.o.d -o lib/Support/CMakeFiles/LLVMSupport.dir/APFloat.cpp.o -c /home/user5/tinygo/llvm-project/llvm/lib/Support/APFloat.cpp
In file included from /home/user5/tinygo/llvm-project/llvm/lib/Support/APFloat.cpp:14:
In file included from /home/user5/tinygo/llvm-project/llvm/include/llvm/ADT/APFloat.h:19:
In file included from /home/user5/tinygo/llvm-project/llvm/include/llvm/ADT/ArrayRef.h:15:
In file included from /home/user5/tinygo/llvm-project/llvm/include/llvm/ADT/STLExtras.h:20:
In file included from /home/user5/tinygo/llvm-project/llvm/include/llvm/ADT/Optional.h:21:
/home/user5/tinygo/llvm-project/llvm/include/llvm/ADT/STLForwardCompat.h:53:22: error: default initialization of an object of const type 'const llvm::in_place_t' without a user-provided default constructor
constexpr in_place_t in_place; // NOLINT(readability-identifier-naming)
                     ^
                             {}
1 error generated.
[12/3024] Building CXX object lib/Demangle/CMakeFiles/LLVMDemangle.dir/ItaniumDemangle.cpp.o
ninja: build stopped: subcommand failed.
Makefile:249: recipe for target 'llvm-build' failed
make: *** [llvm-build] Error 1
```

Additional information: 
Operating System: Ubuntu 16.04.7 LTS
Kernel: Linux 4.4.0-190-generic
Architecture: x86-64

Any pointers would be helpful.Thank you.

aykevl, 2023-02-10:
Actually, it looks like you're using Clang (not GCC). What's the output of `clang --version`?

Sunidhi-Gaonkar1, 2023-02-13:
Thank you for the quick response and sorry for the confusion.
This is related to https://github.com/tinygo-org/tinygo/issues/3041.
I am trying to build binaries for linux/ppc64le by modifying Github actions workflows and cross compilation done similar to arm and arm64 Linux builds.

> Why do you install powerpc64le-linux-gnu?

I have installed powerpc64le-linux-gnu as it provides packages required for cross-compilation.

Below is the output for `gcc --version` and `clang --version`

```
$ gcc --version
gcc (Ubuntu 7.3.0-23ubuntu2~16.04.york0) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.

$ clang --version
clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

aykevl, 2023-02-13:
Can you describe _exactly_ which steps you took? Including all commands?
I see that you are building using Clang, so it appears you have modified `CC` at least.

davecheney, 2026-10-24:
Thanks for the report. If the build instructions above still fail with current TinyGo and its required compiler versions, please open a new issue with the commands and full error output.

closed 2026-10-24 by davecheney