tinygo-org/tinygo #3440: Github contribution - inexperience user

Gustavomurta, 2023-02-11:
I'm have no experience with github. And I wish to contribute.  I actually think it could be simpler.
Anybody can help me? 

My project - **ROM Reader with Arduino Mega (wiht Golang and Tinygo)**
https://github.com/Gustavomurta/Arduino_Mega_Programmer/tree/main/ROM%20Reader%20with%20Arduino

I want to extend this project later. 

Thanks in advance.

soypat, 2023-02-12:
Hey! You'll get a much better and faster response asking on the tinygo channel in [Gophers slack](https://invite.slack.golangbridge.org/). This is the link to the invite: https://invite.slack.golangbridge.org/

Gustavomurta, 2023-02-13:
Thanks soypat!

deadprogram, 2024-11-30:
Closing since questions appears to have been answered. Thank you!

closed 2024-11-30 by deadprogram