tinygo-org/tinygo #3441: warning: no avr-gcc installation can be found on the system, cannot link standard libraries

tagus, 2023-02-11:
Hello, running into an issue trying to flash the following program onto an arduino uno. 

```go
package main

import (
	"machine"
	"time"

	"github.com/aykevl/ledsgo"
	"tinygo.org/x/drivers/ws2812"
)

func main() {
	var brightness byte = 0x80
	pin := machine.PD6
	np := ws2812.New(pin)

	strip := make(ledsgo.Strip, 16)
	pin.Configure(machine.PinConfig{Mode: machine.PinOutput})

	for {
		now := time.Now().UnixNano()
		for i := range strip {
			color := ledsgo.Color{
				H: uint16(now>>15) - uint16(i)<<12,
				S: 0xff,
				V: brightness,
			}
			strip[i] = color.Spectrum()
		}
		np.WriteColors(strip)
		time.Sleep(time.Second / 100)
	}
}
```

I tried flashing the program using `tinygo flash -target=arduino` only to see the following:

```bash
# tinygo.org/x/drivers/ws2812
warning: no avr-gcc installation can be found on the system, cannot link standard libraries
warning: standard library not linked and so no interrupt vector table or compiler runtime routines will be linked
warning: no avr-gcc installation can be found on the system, cannot link standard libraries
warning: standard library not linked and so no interrupt vector table or compiler runtime routines will be linked
warning: no avr-gcc installation can be found on the system, cannot link standard libraries
warning: standard library not linked and so no interrupt vector table or compiler runtime routines will be linked
```

However I followed the additional [instructions](https://tinygo.org/getting-started/install/macos/) for arduino and ensured that both `avrdude` and `avr-gcc` have been installed.

```bash
> which avr-gcc
/usr/local/bin/avr-gcc
```

```bash
> which avrdude
/usr/local/bin/avrdude
```

Any ideas what might be going on here? I was thinking perhaps something is off with my`PATH` but not totally sure.

aykevl, 2023-02-12:
Can you try the new release? This should have been fixed.

tagus, 2023-02-12:
Thanks for your reply, I tried with the `0.27.0` and seeing a new issue now.

```bash
> tinygo flash -target=arduino
tinygo:ld.lld: error: undefined symbol: __udivmodhi4
>>> referenced by gc_blocks.go:246 (/usr/local/Cellar/tinygo/0.27.0/src/runtime/gc_blocks.go:246)
>>>               /Users/sugat/Library/Caches/tinygo/thinlto/llvmcache-9FF7B98A8CFB015926308FB9F4AC77FBFD497A23:(runtime.run)
>>> did you mean: __udivmodsi4
>>> defined in: /Users/sugat/Library/Caches/tinygo/thinlto/llvmcache-9FF7B98A8CFB015926308FB9F4AC77FBFD497A23
failed to run tool: ld.lld
error: failed to link /var/folders/y7/vfh0h2pn4vd75dgj3hwrjkgw0000gn/T/tinygo3650286500/main: exit status 1
```
I'm on macOS Monterey `12.6.3`, could this be an issue with outdated command line tools?

aykevl, 2023-02-12:
Can you try `tinygo clean`? It might still be using an old cached version of compiler-rt.

> could this be an issue with outdated command line tools?

No, version 0.27.0 doesn't use any command line tools anymore (except for avrdude for flashing).

tagus, 2023-02-12:
that seems to have worked, thanks for your help

closed 2023-02-12 by tagus