tinygo-org/tinygo #3442: ESP32-S3

denysvitali, 2023-02-12:
Add support for [ESP32-S3](https://www.espressif.com/en/products/socs/esp32-s3).

This SOC might require a similar port as with the ESP32-S2 (#3073).

aokhrimenko, 2023-08-27:
any news? I wanted to try Tinygo with my ESP32-S3, and looks like there is still no support for it :(

denysvitali, 2023-10-02:
I've implemented partial support now. Unfortunately I didn't get it to print anything to UART or blink any LED / trigger any GPIO.

The resulting binary doesn't crash, which is a good start I guess.

officer47p, 2025-03-16:
Hey, just found this thing when I wanted to program my xiao-esp32s3.
I see that it's been a while since the last activity on this issue. I was interested in implementing this feature but I'm lost.
So far, found this branch https://github.com/denysvitali/tinygo/tree/feature/add-esp32s3 and plan to spend some time understanding it in the weekend.

Is there anything that either @deadprogram or @denysvitali knows that can be shared here to continue this work?

denysvitali, 2025-03-17:
I gave a presentation about this topic a while back. Hopefully it helps:

https://docs.google.com/presentation/d/18jWccV-F2FguZiB5gXLkQFAhUFK_yl7FwkgtldwstxI/edit?usp=drivesdk

eerison, 2025-04-13:
no success to make this work :(

denysvitali, 2025-04-15:
@eerison Where are you stuck?

ralfuslongus, 2025-11-03:
Very interested in this target too. Hope you can make it

officer47p, 2025-11-03:
TinyGo can be huge, but unfortunately, it appears that the maintainers are busy with their own projects.
There is limited information online to help developers get started on the project, as it requires some prior knowledge that is not widely available.

closed 2025-12-17 by QuLogic