tinygo-org/tinygo #3444: Invalid DWARF sections are produced in Wasm targets

mathetake, 2023-02-13:
It looks like a DWARF sections in Wasm binaries are invalid or slightly incorrect since the beginning.
As you can see in the following repro step, wasmtime failed to produce the DWARF stack trace even with `export WASMTIME_BACKTRACE_DETAILS=1` option.

```bash
$ cat main.go

package main

func main() {
	a()
}

func a() {
	b()
}

func b() {
	c()
}

func c() {
	panic("NOOOOOOOOOOOOOOO")
}

$ tinygo build -target wasi -o main.wasm main.go
$ wasm-tools objdump main.wasm | grep .debug
  custom ".debug_info"                   |     0x1b1b -     0x443b |     10528 bytes | 1 count
  custom ".debug_pubtypes"               |     0x444e -     0x4853 |      1029 bytes | 1 count
  custom ".debug_loc"                    |     0x4861 -     0x60c4 |      6243 bytes | 1 count
  custom ".debug_ranges"                 |     0x60d5 -     0x63a5 |       720 bytes | 1 count
  custom ".debug_aranges"                |     0x63b6 -     0x63e6 |        48 bytes | 1 count
  custom ".debug_abbrev"                 |     0x63f7 -     0x67eb |      1012 bytes | 1 count
  custom ".debug_line"                   |     0x67fa -     0x7e96 |      5788 bytes | 1 count
  custom ".debug_str"                    |     0x7ea4 -     0x8ee8 |      4164 bytes | 1 count
  custom ".debug_pubnames"               |     0x8efb -     0x9bbc |      3265 bytes | 1 count

$ export WASMTIME_BACKTRACE_DETAILS=1
$ wasmtime run main.wasm 
panic: NOOOOOOOOOOOOOOO
Error: failed to run main module `main.wasm`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0: 0x17d6 - <unknown>!runtime._panic
           1: 0x1a05 - <unknown>!char
           2: 0x19f5 - <unknown>!bool
           3: 0x19eb - <unknown>!signed char
           4: 0x19e1 - <unknown>!main.main
           5: 0x19c0 - <unknown>!runtime.run
           6: 0x19aa - <unknown>!_start
    2: wasm trap: wasm `unreachable` instruction executed
```

### Affected versions

I verified the same goes for 0.24, 0.26 and 0.27 of TinyGo.

### Note

- I am not sure if this is limited to Wasm targets. 
- Not only wasmtime, but also wazero cannot correctly produce the backtrace, though it might be possible both of runtimes are wrong in the DWARF implementation. As for the reference, wazero produces the following trace which contains with the weird repetition of `/usr/local/tinygo/src/runtime/panic.go:59:7`.
```
$ wazero run main_027.wasm     
panic: NOOOOOOOOOOOOOOO
error instantiating wasm binary: module[] function[_start] failed: wasm error: unreachable
wasm stack trace:
        .runtime._panic(i32)
                0x1919: /usr/local/tinygo/src/runtime/runtime_tinygowasm.go:72:6 (inlined)
                        /usr/local/tinygo/src/runtime/panic.go:59:7
        .c()
                0x2fa9: /tmp/wazero/main.go:19:7 (inlined)
        .b()
                0x2f47: /tmp/wazero/main.go:14:3 (inlined)
        .a()
                0x2ee9: /tmp/wazero/main.go:9:3 (inlined)
        .main.main()
                0x20cd: /tmp/wazero/main.go:4:3 (inlined)
        .runtime.run$1()
                0x1ffa: /usr/local/tinygo/src/runtime/scheduler_any.go:25:11 (inlined)
                        /usr/local/tinygo/src/runtime/panic.go:59:7
        .runtime.run$1$gowrapper(i32)
                0x6f6: /usr/local/tinygo/src/runtime/scheduler_any.go:23:2 (inlined)
                       /usr/local/tinygo/src/runtime/gc_stack_portable.go:46:21
        .tinygo_launch(i32)
                0x23: /usr/local/tinygo/src/internal/task/task_asyncify_wasm.S:59
        .runtime.scheduler()
                0x1ef2: /usr/local/tinygo/src/internal/task/task_asyncify.go:109:17 (inlined)
                        /usr/local/tinygo/src/runtime/panic.go:59:7
        .runtime.run()
                0x1db8: /usr/local/tinygo/src/runtime/scheduler_any.go:28:11 (inlined)
                        /usr/local/tinygo/src/runtime/panic.go:59:7
        ._start()
                0x1d38: /usr/local/tinygo/src/runtime/runtime_wasm_wasi.go:21:5 (inlined)
                        /usr/local/tinygo/src/runtime/panic.go:59:7

```

- For reference, the Zig code with exactly the same logic can produce the correct stack trace:

```bash
$ wasmtime run ../zig/main.wasm
Error: failed to run main module `../zig/main.wasm`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0:   0x7d - builtin.default_panic
                           at /Users/adrian/Downloads/zig-macos-x86_64-0.11.0-dev.1499+23b7d2889/lib/std/builtin.zig:858:17
           1:   0xa6 - main.inlined_b
                           at /Users/adrian/oss/wazero/internal/testing/dwarftestdata/testdata/zig/main.zig:10:5              - main.inlined_a
                           at /Users/adrian/oss/wazero/internal/testing/dwarftestdata/testdata/zig/main.zig:6:5              - main.main
                           at /Users/adrian/oss/wazero/internal/testing/dwarftestdata/testdata/zig/main.zig:2:5
           2:   0xb0 - start.callMain
                           at /Users/adrian/Downloads/zig-macos-x86_64-0.11.0-dev.1499+23b7d2889/lib/std/start.zig:616:37              - _start
                           at /Users/adrian/Downloads/zig-macos-x86_64-0.11.0-dev.1499+23b7d2889/lib/std/start.zig:232:5
    2: wasm trap: wasm `unreachable` instruction executed
```

mathetake, 2023-02-13:
cc @dgryski @anuraaga

fgsch, 2023-02-13:
Works as expected with `-opt 0` so it looks like the optimisations are affecting this:

```
panic: NOOOOOOOOOOOOOOO
Error: failed to run main module `main.wasm`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0: 0x71ef - runtime.abort
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/runtime_tinygowasm.go:72:6
           1: 0xa6e8 - runtime._panic
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/panic.go:52:7
           2: 0x1c92c - main.c
                           at /tmp/b/main.go:16:7
           3: 0x1c82f - main.b
                           at /tmp/b/main.go:12:3
           4: 0x1c721 - main.a
                           at /tmp/b/main.go:8:3
           5: 0xdd22 - main.main
                           at /tmp/b/main.go:4:3
           6: 0xd17b - runtime.run$1
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/scheduler_any.go:25:11
           7: 0xb839 - <goroutine wrapper>
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/scheduler_any.go:23:2
           8:  0x2c7 - tinygo_launch
                           at /usr/local/Cellar/tinygo/0.27.0/src/internal/task/task_asyncify_wasm.S:59
           9: 0x18ce - (*internal/task.Task).Resume
                           at /usr/local/Cellar/tinygo/0.27.0/src/internal/task/task_asyncify.go:109:17
          10: 0xc9d2 - runtime.scheduler
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/scheduler.go:236:11
          11: 0xb710 - runtime.run
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/scheduler_any.go:28:11
          12: 0xb5b7 - _start
                           at /usr/local/Cellar/tinygo/0.27.0/src/runtime/runtime_wasm_wasi.go:21:5
    2: wasm trap: wasm `unreachable` instruction executed
```

aykevl, 2023-02-15:
There are many possible causes for this issue:
- `wasm-opt` might be messing with DWARF in a non-compatible way.
- LLVM currently breaks debug information a lot, especially at `-O2`. This is also an issue for Clang for example.
- TinyGo might not emit correct debug information (unlikely because `-opt=0` does work, as shown above).

dgryski, 2023-05-18:
Is this still broken?
```
~/go/src/github.com/dgryski/bug/pan $ tinygo version
tinygo version 0.28.0-dev-e41f2915 darwin/amd64 (using go version go1.20.3 and LLVM version 15.0.0)
~/go/src/github.com/dgryski/bug/pan $ tinygo run -target=wasi main.go
panic: NOOOOOOOOOOOOOOO
Error: failed to run main module `/var/folders/5b/_hr1d1fd3qn4t9vtfx5p61wm0000gp/T/tinygo152240676/main`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0: 0x1a62 - runtime.abort
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/runtime_tinygowasm.go:70:6              - runtime._panic
                           at /Users/dgryski/go/src/github.com/tinygo-org/tinygo/src/runtime/panic.go:52:7
           1: 0x3168 - main.c
                           at /Users/dgryski/go/src/github.com/dgryski/bug/pan/main.go:16:7
           2: 0x3106 - main.b
                           at /Users/dgryski/go/src/github.com/dgryski/bug/pan/main.go:12:3
           3: 0x30a8 - main.a
                           at /Users/dgryski/go/src/github.com/dgryski/bug/pan/main.go:8:3
           4: 0x22b8 - main.main
                           at /Users/dgryski/go/src/github.com/dgryski/bug/pan/main.go:4:3
```

mathetake, 2023-05-21:
OK verified it's started working correctly with the lastest wasmtime. Closing!

closed 2023-05-21 by mathetake