tinygo-org/tinygo #3450: tinygo 0.27 cannot compile to GOOS=darwin on arm64

gtsteffaniak, 2023-02-15:
Anytime I try to add `GOOD=darwin` during compile-time, I get the following error:

```
GOOS=darwin tinygo build .
# runtime
/usr/local/lib/tinygo/src/runtime/scheduler.go:22:24: undefined: GOOS
/usr/local/lib/tinygo/src/runtime/runtime_unix.go:181:17: undefined: clock_MONOTONIC_RAW
/usr/local/lib/tinygo/src/runtime/runtime_unix.go:191:16: undefined: clock_REALTIME
/usr/local/lib/tinygo/src/runtime/runtime_unix.go:225:34: undefined: flag_PROT_READ
/usr/local/lib/tinygo/src/runtime/runtime_unix.go:225:49: undefined: flag_PROT_WRITE
/usr/local/lib/tinygo/src/runtime/runtime_unix.go:225:66: undefined: flag_MAP_PRIVATE
/usr/local/lib/tinygo/src/runtime/runtime_unix.go:225:83: undefined: flag_MAP_ANONYMOUS
```

- Using tinygo_0.27.0_arm64.deb on linux docker image (host machine is m1 macbook).
- Also doesn't work when using amd64 linux image.
- WORKS with GOOS=linux and GOOS=windows

aykevl, 2023-02-15:
I can't reproduce this in a Debian LXC container (on arm64).
I suspect something went wrong with your installation. Can you share the output of `which tinygo` and `tinygo env`?

dgryski, 2023-02-15:
I develop entirely on Darwin and I also don't have this problem. Definitely something with the install.

tristanmorgan, 2024-01-22:
I'm getting the same error with the docker container [tinygo/tinygo:0.30.0](https://hub.docker.com/layers/tinygo/tinygo/0.30.0/images/sha256-441e37fc9ba1f0f4f14d0cbf81e3560c561837497b135272b4ea800e3b57174c), I am trying to cross compile by exporting GOOS=darwin and I get the error.

```bash
$ tinygo env
GOOS="darwin"
GOARCH="amd64"
GOROOT="/usr/local/go"
GOPATH="/home/tinygo/go"
GOCACHE="/home/tinygo/.cache/tinygo"
CGO_ENABLED="0"
TINYGOROOT="/usr/local/tinygo"
```

tristanmorgan, 2024-01-22:
Hi @aykevl,

The steps I took were: I created an empty folder with a basic hello world program.
```golang
package main

import "fmt"

func main() {
    fmt.Println("Hello, World!")
}
```

and 

```golang
module example.com/test

go 1.21.1
```
then ran the following to get an interactive prompt in the docker container.
`docker run --rm -it -v ${PWD}:/src tinygo/tinygo:0.30.0 /bin/sh`

next inside the container I exported environment variables and tried to compile.
```bash
$ cd /src
$ export GOOS=darwin
$ export GOARCH=arm64
$ export CGO_ENABLED=0
$ tinygo env
GOOS="darwin"
GOARCH="arm64"
GOROOT="/usr/local/go"
GOPATH="/home/tinygo/go"
GOCACHE="/home/tinygo/.cache/tinygo"
CGO_ENABLED="0"
TINYGOROOT="/usr/local/tinygo"
$ tinygo build -o test
# runtime
/usr/local/tinygo/src/runtime/scheduler.go:22:24: undefined: GOOS
/usr/local/tinygo/src/runtime/gc_blocks.go:423:2: undefined: findGlobals
/usr/local/tinygo/src/runtime/runtime_unix.go:181:17: undefined: clock_MONOTONIC_RAW
/usr/local/tinygo/src/runtime/runtime_unix.go:191:16: undefined: clock_REALTIME
/usr/local/tinygo/src/runtime/runtime_unix.go:225:34: undefined: flag_PROT_READ
/usr/local/tinygo/src/runtime/runtime_unix.go:225:49: undefined: flag_PROT_WRITE
/usr/local/tinygo/src/runtime/runtime_unix.go:225:66: undefined: flag_MAP_PRIVATE
/usr/local/tinygo/src/runtime/runtime_unix.go:225:83: undefined: flag_MAP_ANONYMOUS
$ exit
```

What does seem to make a big difference is `CGO_ENABLED=0`, if CGO is enabled it will compile correctly, disabled it breaks.

tristanmorgan, 2024-01-22:
When I try a slightly more complex program with CGO enabled it hits the error in #3907 
```bash
$ tinygo build -o tree_darwin
# os/user
/usr/local/go/src/os/user/cgo_lookup_unix.go:177:30: undefined: syscall.ERANGE
```

aykevl, 2024-02-19:
> What does seem to make a big difference is `CGO_ENABLED=0`, if CGO is enabled it will compile correctly, disabled it breaks.

Ah, so this is the issue. TinyGo requires CGo and will break for some targets if it isn't enabled.
I've made a PR to fix this here: https://github.com/tinygo-org/tinygo/pull/4136
In the meantime, make sure you use `CGO_ENABLED=1` for TinyGo.

Closing because while this is certainly confusing, it's not necessarily a bug and I've made a PR to avoid these kinds of issues in the future.

closed 2024-02-19 by aykevl