tinygo-org/tinygo #3454: Go test for jfreymuth/oggvorbis fails

sago35, 2023-02-16:
The test was successful for `tinygo 0.25`.
However, `TinyGo 0.26` or later fails.

This is caused by oggvorbis.NewReader() returning an error.
This prevents normal use as well as testing.
It was working in TinyGo 0.25.

https://github.com/jfreymuth/oggvorbis

```
$ /c/tinygo/tinygo0.25.0.windows-amd64/tinygo/bin/tinygo test .
ld.lld: warning: duplicate /export option: hypot
ld.lld: warning: duplicate /export option: nextafter
ok      github.com/jfreymuth/oggvorbis  0.121s
```

```
$ /c/tinygo/tinygo0.26.0.windows-amd64/tinygo/bin/tinygo test .
ld.lld: warning: duplicate /export option: hypot
ld.lld: warning: duplicate /export option: nextafter
panic: runtime error: nil pointer dereference
FAIL    github.com/jfreymuth/oggvorbis  0.186s
FAIL
```

```
$ /c/tinygo/tinygo0.27.0.windows-amd64/tinygo/bin/tinygo test .
FAIL    github.com/jfreymuth/oggvorbis  0.209s
```

```
$ go test .
ok      github.com/jfreymuth/oggvorbis  (cached)
```

dgryski, 2023-02-16:
Weird that it started failing with newer versions.  I'll take a look and investigate.

dgryski, 2023-02-16:
First pass investigation might blame the `Seek()` system calls.   Leaving for now..

aykevl, 2023-02-16:
Ok, so the nil pointer panic is caused because `oggvorbis.NewReader` (in `TestLengthForUnexpectedEof`) returns an error, this is logged using `t.Fatal`, but we don't actually stop at `t.Fatal`. I didn't investigate further.

dgryski, 2023-02-17:
Yeah, the issue though is that `NewReader` shouldn't be returning an error in the middle of the test, and my thought is because the Seek() behaviour to rewind the ogg stream to the beginning isn't working, so the packet decoding step fails.

dgryski, 2023-03-01:
This is a bug in the upstream floating point code: https://github.com/tinygo-org/tinygo/issues/3482#issuecomment-1449280190

dgryski, 2023-03-01:
Filed https://github.com/jfreymuth/vorbis/issues/11

closed 2023-03-01 by dgryski