tinygo-org/tinygo #3456: Timer.Reset() not implemented

jadefox10200, 2023-02-17:
timer.modTimer appears to not be implemented. 

The error I am getting is: 
```tinygo:ld.lld: error: undefined symbol: time.modTimer
referenced by tick.go:60 (/opt/homebrew/Cellar/go/1.19.4/libexec/src/time/tick.go:60)
/Users/stillshot/Library/Caches/tinygo/thinlto/llvmcache-0ACC4295CB94FD749B6046AA38636C6DBE5FF8A5:((*time.Ticker).Reset)
failed to run tool: ld.lld
error: failed to link /var/folders/mn/5m1sf0ys2hv3pypqprmrrdp80000gn/T/tinygo1735139416/main: exit status 1
```
I'm trying to use the function *time.Reset() to reset a ticker but it apparently doesn't like that. This is on an RP2040. tinygo version 0.26.0 darwin/amd64 (using go version go1.19.4 and LLVM version 14.0.0)

huberty89, 2024-03-29:
I got the same for:
```
tinygo version 0.31.2 darwin/arm64 (using go version go1.22.1 and LLVM version 17.0.1)
```
also for RP2040.

It's strange because in previous version of TinyGo it works

tdunning, 2025-02-04:
This issue is not really an enhancement. Existing functionality does not behave according to spec. That's a bug.

Here is more info ported from #4688 

I wrote a small program that was using a Ticker as a timeout and found that resetting the Ticker exposed a missing bit in the runtime.

Here is a tiny program that illustrates this:

```go
package main

import (
	"time"
)

func main() {
	t := time.NewTicker(10 * time.Millisecond)
	t.Reset(time.Duration(100) * time.Millisecond)
}
```

Attempting to compile and flash this results in this output:
```bash
$  tinygo flash -target=pico2 -scheduler=tasks -gc=conservative -size=full -stack-size=20kb -monitor TickerBug/main.go
/usr/local/go/src/time/tick.go:58: linker could not find symbol time.modTimer
$
```

abulka, 2025-08-02:
Does this issue also apply to `time.Timer` and `timer.Reset`? For example I seem to be able to run this code:

```go
        var timer *time.Timer       // pointer to a time.Timer object
	var timerC <-chan time.Time // receive-only channel of type time.Time
	for {
		nextWake := 30 * time.Millisecond

		if timer == nil {
			timer = time.NewTimer(nextWake)
			timerC = timer.C
		} else {
			timer.Stop()
			timer.Reset(nextWake) // Update the timer to the new duration wanted
		}

		select {
		case <-state.edgeEvents:
		case <-timerC:
			print(".")
		}
```

albeit I do get strange bugs after a while, which may or may not be related.

tdunning, 2025-08-03:
I don't know. It's been quite a while and might actually be fixed by now.

tdunning, 2026-10-24:
Has this issue gone stale?

davecheney, 2026-10-24:
`Ticker.Reset` now builds and runs with current TinyGo. The Go 1.23+ timer implementation was added in https://github.com/tinygo-org/tinygo/commit/e865db232bad0a2a5152749edd6a9ce32557c8b7.

This program resets a one-hour ticker to 10 ms and checks that it fires:

```go
package main

import "time"

func main() {
    ticker := time.NewTicker(time.Hour)
    defer ticker.Stop()

    ticker.Reset(10 * time.Millisecond)
    select {
    case <-ticker.C:
        println("Ticker.Reset passed")
    case <-time.After(2 * time.Second):
        panic("ticker did not fire after reset")
    }
}
```

Checked with TinyGo 0.44.0-dev-eae3f875 and Go 1.27.3:

```sh
tinygo run main.go
tinygo run -target=wasm main.go
tinygo build -target=pico -o check-pico.elf main.go
tinygo build -target=pico2 -o check-pico2.elf main.go
```

Both runs print `Ticker.Reset passed`; both board builds succeed. The board builds were compile checks, not hardware tests. Please open a new issue with a minimal reproducer if problems remain.

closed 2026-10-24 by davecheney