tinygo-org/tinygo #3460: Virtual filesystem support broken on baremetal targets

bgould, 2023-02-18:
As best I can tell, virtual filesystem support on baremetal targets was effectively removed by https://github.com/tinygo-org/tinygo/commit/a680bfbb7afcb424c4d8110255336f85fd66e7e6

NewFile in src/os/file_other.go always returns a stdioFileHandle now (https://github.com/tinygo-org/tinygo/commit/a680bfbb7afcb424c4d8110255336f85fd66e7e6#diff-639e6d3c80d353b492953bcbd0b3eec979e6c88a776684acd1ac961b6217e24eR36) and should be refactored to delegate a mounted virtual filesystem if it exists.

deadprogram, 2023-03-11:
Now that we have all these processors with flash support, I think I am feeling the pain of this issue.

```
Connected to /dev/ttyACM0. Press Ctrl-C to exit.
SPI Configured. Reading flash info
==> lsblk

-------------------------------------
 Device Information:  
-------------------------------------
 flash data start: 08015800
 flash data end:  08040000
-------------------------------------

==> format
Successfully formatted LittleFS filesystem.

==> ls
Could not open directory /: littlefs: Corrupted
```

deadprogram, 2023-03-25:
What can I do to help resolve this @bgould ?

deadprogram, 2023-05-20:
However, FAT filesystem is still broken, which was perhaps not clear from my previous comment.

deadprogram, 2023-12-08:
@bgould I need your help to figure out what to fix, and where. :crying_cat_face: 

How is it possible that LittleFS is still working but FatFS not? I am very confused.

soypat, 2024-01-18:
@deadprogram @bgould Heads up to both of y'all, I'm writing a FatFS port in pure Go

soypat, 2026-01-14:
https://github.com/soypat/fat  Pure Go.