tinygo-org/tinygo #3466: Fresh install from source cannot find packages in GOROOT

raoulmillais, 2023-02-19:
I have installed tinygo from source using the instructions on a machine that I have not previously done any go development  on (I.E. no changes to my go env anywhere) It also has the latest go (1.20).

I followed the tutorial and also did:

* `go mod init raoulmillais.com/tinygo-tutorial`
* `go mod tidy`

but I cannot flash the tutorial because it cannot find the tinygo packages in GOROOT.

```
λ ~/embedded/tinygo-tutorial/  » tinygo flash -target=pico main.go
../../.cache/tinygo/goroot-<snip>/src/machine/board_pico.go:6:2: package device/rp is not in GOROOT (/home/raoul/.cache/tinygo/goroot-<snip>/src/device/rp)

```
<details>
  <summary>My `go env`</summary>

```
λ ~/embedded/tinygo-tutorial/  » go env
GO111MODULE=""
GOARCH="amd64"
GOBIN=""
GOCACHE="/home/raoul/.cache/go-build"
GOENV="/home/raoul/.config/go/env"
GOEXE=""
GOEXPERIMENT=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GOINSECURE=""
GOMODCACHE="/home/raoul/go/pkg/mod"
GONOPROXY=""
GONOSUMDB=""
GOOS="linux"
GOPATH="/home/raoul/go"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/usr/lib/go"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/usr/lib/go/pkg/tool/linux_amd64"
GOVCS=""
GOVERSION="go1.20.1"
GCCGO="gccgo"
GOAMD64="v1"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD="/home/raoul/embedded/tinygo-tutorial/go.mod"
GOWORK=""
CGO_CFLAGS="-O2 -g"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-O2 -g"
CGO_FFLAGS="-O2 -g"
CGO_LDFLAGS="-O2 -g"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -Wl,--no-gc-sections -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build306879867=/tmp/go-build -gno-record-gcc-switches"
```
</details>

`tinygo env`:

```
λ ~/embedded/tinygo-tutorial/  » tinygo env
GOOS="linux"
GOARCH="amd64"
GOROOT="/usr/lib/go"
GOPATH="/home/raoul/go"
GOCACHE="/home/raoul/.cache/tinygo"
CGO_ENABLED="1"
TINYGOROOT="/home/raoul/builds/tinygo"
```

I've compared with `go env` and `tinygo env` inside the tinygo docker container and I can't spot any differences apart from the paths. 

Creating a regular go module and importing std libraries and 3rd party libraries works just fine.

Shouldn't this just work with fresh installs of both go and tinygo?  Any help would be much appreciated.

aykevl, 2023-02-19:
You need to run `make gen-device` to fix this.

The build from source instructions do mention this, although it's admittedly easy to miss: https://tinygo.org/docs/guides/build/#additional-requirements

raoulmillais, 2023-02-19:
Ah brilliant, that worked.  Thanks

closed 2023-02-19 by raoulmillais