tinygo-org/tinygo #3474: USB: Composite HID Devices

neildavis, 2023-02-22:
Firstly,  should [`joystick`](https://github.com/tinygo-org/tinygo/tree/release/src/machine/usb/joystick) be located under [`machine/usb/hid`](https://github.com/tinygo-org/tinygo/tree/release/src/machine/usb/hid)  (along with [`keyboard`](https://github.com/tinygo-org/tinygo/tree/release/src/machine/usb/hid/keyboard) & [`mouse`](https://github.com/tinygo-org/tinygo/tree/release/src/machine/usb/hid/mouse)) instead of in `machine/usb` (with siblings [`cdc`](https://github.com/tinygo-org/tinygo/tree/release/src/machine/usb/cdc) & [`midi`](https://github.com/tinygo-org/tinygo/tree/release/src/machine/usb/midi))? As I understand, CDC, MIDI & HID are all different USB 'device classes' whereas `joystick` is [clearly](https://github.com/tinygo-org/tinygo/blob/v0.27.0/src/machine/usb/joystick/joystick.go#L46) [using HID](https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/src/machine/usb.go#L277)

Secondly, (and more importantly) it appears that more than one of `mouse`, `keyboard` & `joystick` cannot be used in a 'composite HID device'. Whist `mouse` & `keyboard` can coexist, only one device actually works and adding `joystick` breaks everything. e.g. the program below will not work as written with the following observations:

1. Whilst the mouse & keyboard show up in Linux as `/dev/input/eventX` & `/dev/input/mouse1`, no events are received by `evtest` on these devices.
2. No event device or joystick device (e.g. `/dev/input/js0`) is created in Linux for the joystick.
3. If the joystick code is removed, then the mouse & keyboard devices are both created but **only mouse events** are received by `evtest`

Using: `tinygo version 0.28.0-dev-1065f06 linux/amd64 (using go version go1.20.1 and LLVM version 15.0.0)` from [this build run](https://github.com/tinygo-org/tinygo/actions/runs/4223759755) - latest available commit on `dev` at time of writing:

```go
package main

import (
	"machine/usb/hid/keyboard"
	"machine/usb/hid/mouse"
	"machine/usb/joystick"
	"time"
)

// Entry point
func main() {
	kb := keyboard.Port()
	ms := mouse.Port()
	js := joystick.Port()

	ticker := time.NewTicker(time.Second)
	cnt := 0
	for range ticker.C {
		x := 100
		button := cnt%2 > 0
		if button {
		}
		js.SetButton(2, button)
		js.SetButton(3, !button)
		js.SetAxis(0, int(x))
		js.SendState()

		if button {
			x = -100
			kb.Press(keyboard.KeySpace)
			ms.Press(mouse.Left)
		} else {
			kb.Press(keyboard.KeyBackspace)
			ms.Release(mouse.Left)
		}
		ms.Move(x, 0)
		cnt++
	}
}
```

If someone can tell what the issue is and point me in the right direction I may be able to contribute to a fix/enhancement.

Basically, I'm looking for composite HID device support similar to that [offered by CircuitPython](https://learn.adafruit.com/customizing-usb-devices-in-circuitpython/hid-devices)

e.g. in CP I can do this:

```python
import usb_hid
import usb_midi
from adafruit_hid.consumer_control import ConsumerControl
from adafruit_hid.consumer_control_code import ConsumerControlCode
from adafruit_hid.mouse import Mouse
from adafruit_hid.keyboard import Keyboard
from hid_gamepad import Gamepad

GAMEPAD_REPORT_DESCRIPTOR = bytes((
# <snip> Gamepad HID report descriptor ...
))
 
gamepad = usb_hid.Device(
    report_descriptor=GAMEPAD_REPORT_DESCRIPTOR,
    usage_page=0x01,           # Generic Desktop Control
    usage=0x04,                # Joystick
    report_ids=(4,),           # Descriptor uses report ID 4.
    in_report_lengths=(10,),   # This gamepad sends 10 bytes in its report.
    out_report_lengths=(0,),   # It does not receive any reports.
)

# Disable MIDI to save on USB I/O endpoints
usb_midi.disable()
# Enable gamepad, consumer control (volume), keyboard and mouse HID devices
usb_hid.enable(
    ( gamepad, usb_hid.Device.CONSUMER_CONTROL, usb_hid.Device.MOUSE, usb_hid.Device.KEYBOARD )
)
# Mouse
mouse = Mouse(usb_hid.devices)
# Keyboard
kb = Keyboard(usb_hid.devices)
# Gamepad
gp = Gamepad(usb_hid.devices)
# Consumer Control
cc = ConsumerControl(usb_hid.devices)
# some code that uses mouse, kb, gp & cc ...
```

sago35, 2023-02-24:
For now, all USB Descriptors (usb.Descriptor{}) are fixed values.
https://github.com/tinygo-org/tinygo/blob/v0.27.0/src/machine/usb/descriptor.go

To dynamically create a composite device, usb.Descriptor must be generated programmatically.
If we actually create a Descriptor for the composite device and check it, we will know what we need to do.
It really helped me when you showed me the descriptor when mediakey support.

Of course, we can check the implementation of tinyusb and others.
https://github.com/hathach/tinyusb

neildavis, 2023-03-01:
Thanks for the pointer @sago35. When I get some free time I will look into this. I'm quite familiar with HID usage & report descriptors, but I will need to read up some more on the higher level ones (device, configuration, interface ... ) to consider how a custom USB config API will look like.

deadprogram, 2023-03-21:
> Firstly, should joystick be located under machine/usb/hid (along with keyboard & mouse) instead of in machine/usb

Yes, please! That will be an excellent start.