tinygo-org/tinygo #3479: Build error on Windows 10

hgmmym, 2023-02-24:
I try to build example blinky1 from tinygo/src and I got this:
```
C:\tinygo\src\examples\blinky1>tinygo version
tinygo version 0.27.0 windows/amd64 (using go version go1.19.4 and LLVM version 15.0.0)

C:\tinygo\src\examples\blinky1>tinygo build -o output ./blinky1.go
ld.lld: error: undefined symbol: __tinygo_gpio_configure
>>> referenced by main
>>>               C:\Users\ADMINI~1\AppData\Local\Temp\tinygo2300293750\main.o

ld.lld: error: undefined symbol: __tinygo_gpio_set
>>> referenced by main
>>>               C:\Users\ADMINI~1\AppData\Local\Temp\tinygo2300293750\main.o
failed to run tool: ld.lld
error: failed to link C:\Users\ADMINI~1\AppData\Local\Temp\tinygo2300293750\main.exe: exit status 1
```

What is that mean?

sago35, 2023-02-24:
try this.

tinygo clean

sago35, 2023-02-24:
You should specify target.

tinygo build -target arduino ...

hgmmym, 2023-02-24:
@sago35 You are GOD D@MN RIGHT! It's work! 
Thanks so much.
This issue should be close, I'm going to enjoy my SBC now.

closed 2023-02-24 by hgmmym