tinygo-org/tinygo #3485: Should additional use of wasm-opt shrink my binary?

nickwesselman, 2023-02-25:
I thought that with wasm-opt integrated into TinyGo, I shouldn't need to invoke it separately to get the smallest possible binary. But it appears to still be shrinking my wasm, and not insignificantly. Is there something different about the wasm-opt invocation from tinygo?

```
tinygo build -target wasi -scheduler=none -gc=leaking -opt=z -no-debug -o build/order-discount-golang-unopt.wasm ./main
wasm-opt -Oz -o build/order-discount-golang.wasm build/order-discount-golang-unopt.wasm
```

```
ls -l build
-rwxr-xr-x  2 nickwesselman  staff  214246 Feb 25 12:25 order-discount-golang-unopt.wasm
-rw-r--r--  2 nickwesselman  staff  196027 Feb 25 12:25 order-discount-golang.wasm
```

dgryski, 2023-02-25:
TinyGo runs `wasm-opt` with some default settings. You can probably get additional benefits by running it again with more aggressive flags.

See for example https://github.com/WebAssembly/binaryen/wiki/Optimizer-Cookbook

nickwesselman, 2023-02-25:
Would be nice if the `opt` flag provided to tinygo was also mapped into `wasm-opt`!

dgryski, 2023-02-25:
It is, but not with all the extra flags; https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/builder/build.go#L812

nickwesselman, 2023-02-25:
So that's my confusion -- in my example above, I use `-opt=z` with tinygo, then only `-Oz` with wasm-opt, and it still reduces the binary size.

dgryski, 2023-02-25:
What version of tinygo are you using? `wasm-opt` was only run when needed previously, and with `-scheduler=none` it wouldn't have been run at all.  Also the wiki page I linked to above does indicate that multiple runs with `-Oz` can make a difference.  That is why for people who are concerned about wasm binary size, they should set up their own build pipeline with a custom `wasm-opt` command.  TinyGo can't please everyone.

nickwesselman, 2023-02-25:
I'm on 0.27.0.

> That is why for people who are concerned about wasm binary size, they should set up their own build pipeline with a custom wasm-opt command

Understood, thanks!

closed 2023-02-25 by nickwesselman