tinygo-org/tinygo #3490: Wasm building fails on Windows 10

BugFreeSoftware, 2023-02-27:
TinyGo crashes on Windows 10 when trying to build Wasm code.
Whether or not I provide a source file to compile or any other flags does not matter.
This command will reproduce the error:

tinygo build -target wasm 

The result is always this error message:
       
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x98 pc=0x7ff6183054dd]

goroutine 1 [running]:
github.com/tinygo-org/tinygo/loader.(*Program).MainPkg(...)
        D:/a/tinygo/tinygo/loader/loader.go:295
github.com/tinygo-org/tinygo/builder.Build({0x7ff61ca238e0, 0x1}, {0x0, 0x0}, {0xc0000a27b0, 0x17}, 0xc0001827e0)
        D:/a/tinygo/tinygo/builder/build.go:194 +0x119d
main.Build({0x7ff61ca238e0, 0x1}, {0x0, 0x0}, 0xc00018e000)
        D:/a/tinygo/tinygo/main.go:168 +0x26f
main.main()
        D:/a/tinygo/tinygo/main.go:1525 +0x3d3f

BugFreeSoftware, 2023-02-28:
Best guess for the error after a quick look is that the MainPkg() function fails on Windows.
Since this function is ALWAYS called no matter the Wasm-specific parameters this would result in the observed behavior.
As an aside, if the previous call returns an error, the code there could crash horribly unless the returned Program actually exists. The error return value is not checked until AFTER the returned value has been used several times already.
See the code that calls MainPkg() here:
https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/builder/build.go#L193

aykevl, 2023-02-28:
Can you provide steps to reproduce this issue?

BugFreeSoftware, 2023-02-28:
I provided the steps in the issue.
Step 1: install TinyGo 0.27.0 on my Windows 10 machine
            (removed 0.26.0 folder, extracted 0.27.0 to exact same place,
just like I have been doing
             for the past 2 years of using and upgrading TinyGo to build my
Wasm code)
Step 2: tinygo build -target wasm <anything here>
            >>>CRASH<<<

>tinygo version
tinygo version 0.27.0 windows/amd64 (using go version go1.20.1 and LLVM
version 15.0.0)

>tinygo env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:/Program Files/Go"
GOPATH="C:\\Users\\skype\\go"
GOCACHE="C:\\Users\\skype\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\tinygo"

>tinygo build
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x98 pc=0x7ff6ebd654dd]

goroutine 1 [running]:
github.com/tinygo-org/tinygo/loader.(*Program).MainPkg(...)
        D:/a/tinygo/tinygo/loader/loader.go:295
github.com/tinygo-org/tinygo/builder.Build({0x7ff6f04838e0, 0x1}, {0x0,
0x0}, {0xc00001c5b8, 0x17}, 0xc0001c6600)
        D:/a/tinygo/tinygo/builder/build.go:194 +0x119d
main.Build({0x7ff6f04838e0, 0x1}, {0x0, 0x0}, 0xc000002180)
        D:/a/tinygo/tinygo/main.go:168 +0x26f
main.main()
        D:/a/tinygo/tinygo/main.go:1525 +0x3d3f

aykevl, 2023-02-28:
Part of being able to reproduce this issue is having the source code. Can you provide some source code that reproduces this issue?

BugFreeSoftware, 2023-02-28:
Hi Ayke,

I noticed the GOCACHE in my previous response, so I cleared the
AppData/Local/tinygo folder, rebooted my machine for good measure, and also
cleared the GoLand caches.
I did get the error once more after that, but now it seems to have vanished.

I will add a remark to the issue and close it.

There is still a little bug in the code, though, because the error return
value is not being checked here, and the subsequent code could continue to
run with a nil value for lprogram.
https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/builder/build.go#L190

Eric.

BugFreeSoftware, 2023-02-28:
Clearing the /AppData/Local/tinygo folder and rebooting my machine seems to have fixed it. 
The error happened once immediately after this action but since has vanished and everything seems to work now. Closing this issue for now.

closed 2023-02-28 by BugFreeSoftware