tinygo-org/tinygo #3494: add new target: cryptoauth trust platform

hugolgst, 2023-02-27:
I would love to see the CryptoAuth Trust Platform being added.
It uses ATSAMD21E18A which is already implemented.
https://www.microchip.com/en-us/development-tool/DM320118

hugolgst, 2023-03-06:
I can't get this code to blink the LED. Any idea why? 
[image]
Here are some information about the chip: https://ww1.microchip.com/downloads/aemDocuments/documents/SCBU/ProductDocuments/BoardDesignFiles/DM320118-schematics.PDF

conejoninja, 2023-03-06:
I think it should be something like this instead:

```go
var led = machine.LED
led.Configure(machine.PinConfig{Mode: machine.PinOutput})
```

hugolgst, 2023-03-06:
Yes but since it is a custom board I did not set PinOutput in the board's file yet.

conejoninja, 2023-03-06:
Ummm, but no need to re-define the PinOutput, since it's already in the `machine_atsamd21.go` file https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_atsamd21.go#L29 plus it should be `11` not `0`

Also, let's try with 
```go
var led = machine.LED
```
instead of 

```go
var led machine.Pin = 2
```

hugolgst, 2023-03-06:
right, I was confused with something else. Now I am indeed using LED and PinOutput, it was just this. I can't thank you enough for your help. Better read the documentation well next time. 

I will try to clean out everything and make a pull request to add my board.

Have a nice day

conejoninja, 2023-03-06:
No problem!
Adding a new board should be easy as long as the chip is already supported like this case.

I proceed to close this issue, please open a PR when your new board is ready!

Thanks for your work

closed 2023-03-06 by conejoninja

hugolgst, 2023-03-07:
https://github.com/tinygo-org/tinygo/pull/3527