tinygo-org/tinygo #3517: `arduino-nano` can't run the `blinky` example on a fresh install.

zealws, 2023-03-05:
Fresh install of `tinygo` on Arch Linux using the official package, but it doesn't seem to be installed correctly.

I'm trying to flash an arduino-nano with the [Blinking LED](https://tinygo.org/docs/tutorials/blinky/) tutorial and it doesn't seem like the `machine` package is found:
```
# tinygo flash -target arduino-nano
# blinky
main.go:6:9: undefined: machine
main.go:7:16: undefined: machine
main.go:8:9: undefined: machine
```

It definitely seems like the `machine` package is present in the GOROOT though:
```
# tinygo info -target arduino-nano
LLVM triple:       avr
GOOS:              linux
GOARCH:            arm
build tags:        avr baremetal linux arm atmega328p atmega avr5 arduino_nano tinygo math_big_pure_go gc.conservative scheduler.none serial.uart
garbage collector: conservative
scheduler:         none
cached GOROOT:     /home/zeal/.cache/tinygo/goroot-50390eaca9e93f981c91237344b3527e23ad8f84acfdb27879b77b65c813de45

# ls -ld $(realpath /home/zeal/.cache/tinygo/goroot-50390eaca9e93f981c91237344b3527e23ad8f84acfdb27879b77b65c813de45/src/machine)
drwxr-xr-x 1 root root 8904 Mar  4 22:28 /usr/lib/tinygo/src/machine
```

Some additional version info:
```
# tinygo version
tinygo version 0.27.0 linux/amd64 (using go version go1.20.1 and LLVM version 15.0.0)

# tinygo env GOROOT
/usr/lib/go
```

Other things I tried that result in the same error:
- Using `go1.19` instead of `go1.20`
- Installing `tinygo` from source on the `release` branch
- Uninstalling the OS package entirely and only using the from-source install
- Running `tinygo clean`
- Manually deleting `$HOME/.cache/tinygo`

I'd suspect an issue with the package, but if I can't run the hello-world example with tinygo from source, I think there's definitely an issue with tinygo itself.

aykevl, 2023-03-05:
This is most likely a broken install. We have a ton of tests in CI that import the machine package, if that weren't possible we certainly would have noticed it.

Can you try installing the binary release from https://github.com/tinygo-org/tinygo/releases to see whether the issue is also present there?

zealws, 2023-03-05:
Yeah, I also suspect an install issue, not a code issue, but I've tried everything I can think of to fix it.

Fwiw, I already tried removing the OS install and installing tinygo from source and the issue persisted, so it's not specific to the OS package.

Same issue using the github release binary as well:
```
# wget https://github.com/tinygo-org/tinygo/releases/download/v0.27.0/tinygo0.27.0.linux-amd64.tar.gz
...

# sha1sum tinygo0.27.0.linux-amd64.tar.gz 
c80eb2536c99dfcc5c1b88789243c212b038e094  tinygo0.27.0.linux-amd64.tar.gz

# tar xzf tinygo0.27.0.linux-amd64.tar.gz 

# tinygo/bin/tinygo version
tinygo version 0.27.0 linux/amd64 (using go version go1.20.1 and LLVM version 15.0.0)

# tinygo/bin/tinygo flash -target arduino-nano main.go 
# command-line-arguments
main.go:6:9: undefined: machine
main.go:7:16: undefined: machine
main.go:8:9: undefined: machine
```

zealws, 2023-03-05:
Same results if I use go1.19 as well:

```
# tinygo/bin/tinygo version
tinygo version 0.27.0 linux/amd64 (using go version go1.19.6 and LLVM version 15.0.0)

# tinygo/bin/tinygo flash -target arduino-nano main.go 
# command-line-arguments
main.go:6:9: undefined: machine
main.go:7:16: undefined: machine
main.go:8:9: undefined: machine
```

aykevl, 2023-03-05:
Can you share the source code?

zealws, 2023-03-05:
It's the blinky example:
https://tinygo.org/docs/tutorials/blinky/

aykevl, 2023-03-05:
The error messages show it's not exactly the given source code. Can you double check the code, or just share here what you have?

zealws, 2023-03-05:
I typed it line-by-line, instead of copy/pasting it, so the line numbers don't match one-to-one, but the code is the same:
```
# cat main.go 
package main

import "time"

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{
		Mode: machine.PinOutput,
	})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}
```

zealws, 2023-03-05:
Oh, I'm dumb. It's missing the `import "machine"`.

I'm used to `goimports` filling in the missing imports, but clearly that doesn't work with the pseudo-stdlib `machine` package.

Sorry for the confusion!

zealws, 2023-03-05:
Fwiw, it does compile if I add the missing import.

closed 2023-03-05 by zealws