tinygo-org/tinygo #3530: Building tinygo on Gentoo

paboum, 2023-03-07:
To build `tinygo` on Gentoo with `go install`, I had to run:
```
ln -s /usr/lib/llvm/15/include/llvm-c /usr/include/llvm-c
ln -s /usr/lib/llvm/15/include/llvm /usr/include/llvm
ln -s /usr/lib/llvm/15/include/x86_64-pc-linux-gnu/llvm/ /usr/include/x86_64-pc-linux-gnu/llvm
ln -s /usr/lib/llvm/15/include/clang-c /usr/include/clang-c

ln -s /usr/lib/llvm/15/lib64/libLLVM-15.so /usr/lib/
ln -s /usr/lib/llvm/15/lib64/libclang.so /usr/lib/
```
Might becoming handy for someone, so I'm just pasting it.

Also, going the extra mile, I can think of two possible actions:
- Gentoo could provide an use flag for building `llvm` to provide those in "debian fashion" as well.
- tinygo could provide a build flag like "LLVM_PATH" so I just use it with `/usr/lib/llvm/15/` value and voila.

Perhaps some of these is already implemented, but I didn't find it easily.

paboum, 2023-03-07:
Also: `make llvm-source` is necessary prior to `go install`, otherwise `compiler-rt-builtins` will not be copied into the right place and non-wasm builds will fail. This should yield a warning or something in my opinion, took me some time to resolve it.

deadprogram, 2023-08-18:
SInce the docs change was already merged, I am going to close this issue. Thanks everyone!

closed 2023-08-18 by deadprogram

firelizzard18, 2024-08-15:
@aykevl This works (kind of):
```
CGO_CPPFLAGS="-I/usr/lib/llvm/17/include" go install -tags=llvm17
```

This does not:
```
go install -tags=llvm17
```

I added `-I/usr/lib/llvm/17/include` to `llvm_config_linux_llvm17.go` and to other files but no mater what I try I get a complaint about `#include "llvm-c/Analysis.h"` unless I set CGO_CPPFLAGS.

funkycode, 2025-01-21:
> [@aykevl](https://github.com/aykevl) This works (kind of):
>
> ```
> CGO_CPPFLAGS="-I/usr/lib/llvm/17/include" go install -tags=llvm17
> ```
>
> This does not:
>
> ```
> go install -tags=llvm17
> ```
>
> I added [...]

thanks for the direction what worked for me was:

```bash
CGO_CPPFLAGS="-I/usr/lib/llvm/18/include" CGO_LDFLAGS="-L/usr/lib/llvm/18/lib64 -lclang" go install -tags=llvm18
```