tinygo-org/tinygo #3532: Race condition

aykevl, 2023-03-08:
We have another race condition, this time it appears to be a bit more serious:

```
==================
WARNING: DATA RACE
Read at 0x00c00056e3b8 by goroutine 44:
  go/types.computeInterfaceTypeSet()
      /usr/local/go/src/go/types/typeset.go:155 +0x44
  go/types.(*Interface).typeSet()
      /usr/local/go/src/go/types/interface.go:29 +0x6c4
  go/types.lookupFieldOrMethod()
      /usr/local/go/src/go/types/lookup.go:184 +0x704
  go/types.LookupFieldOrMethod()
      /usr/local/go/src/go/types/lookup.go:66 +0xc0
  golang.org/x/tools/go/ssa.(*Function).selection()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/func.go:68 +0x1a0
  golang.org/x/tools/go/ssa.(*builder).setCallFunc()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:985 +0x100
  golang.org/x/tools/go/ssa.(*builder).setCall()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:1120 +0x44
  golang.org/x/tools/go/ssa.(*builder).expr0()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:704 +0x320c
  golang.org/x/tools/go/ssa.(*builder).expr()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:625 +0x1ac
  golang.org/x/tools/go/ssa.(*builder).stmt()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2160 +0x1e0
  golang.org/x/tools/go/ssa.(*builder).stmtList()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:946 +0x78
  golang.org/x/tools/go/ssa.(*builder).stmt()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2277 +0xc08
  golang.org/x/tools/go/ssa.(*builder).buildFunctionBody()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2391 +0x54c
  golang.org/x/tools/go/ssa.(*builder).buildFunction()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2326 +0x5c
  golang.org/x/tools/go/ssa.(*builder).buildCreated()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2413 +0x48
  golang.org/x/tools/go/ssa.(*Package).build()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2606 +0xfec
  golang.org/x/tools/go/ssa.(*Package).build-fm()
      <autogenerated>:1 +0x34
  sync.(*Once).doSlow()
      /usr/local/go/src/sync/once.go:74 +0xb0
  sync.(*Once).Do()
      /usr/local/go/src/sync/once.go:65 +0x40
  golang.org/x/tools/go/ssa.(*Package).Build()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/ssa/builder.go:2477 +0x2bc
  github.com/tinygo-org/tinygo/compiler.CompilePackage()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:280 +0x294
  github.com/tinygo-org/tinygo/builder.Build.func3()
      /home/ayke/src/tinygo/tinygo/builder/build.go:357 +0x254
  github.com/tinygo-org/tinygo/builder.runJob()
      /home/ayke/src/tinygo/tinygo/builder/jobs.go:222 +0x70
  github.com/tinygo-org/tinygo/builder.runJobs.func2()
      /home/ayke/src/tinygo/tinygo/builder/jobs.go:123 +0x40

Previous write at 0x00c00056e3b8 by goroutine 45:
  go/types.computeInterfaceTypeSet()
      /usr/local/go/src/go/types/typeset.go:194 +0x130
  go/types.(*Interface).typeSet()
      /usr/local/go/src/go/types/interface.go:29 +0x354
  go/types.(*Interface).NumMethods()
      /usr/local/go/src/go/types/interface.go:113 +0x340
  golang.org/x/tools/go/types/typeutil.Hasher.hashFor()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/types/typeutil/map.go:331 +0x37c
  golang.org/x/tools/go/types/typeutil.Hasher.Hash()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/types/typeutil/map.go:239 +0x84
  golang.org/x/tools/go/types/typeutil.(*Map).At()
      /home/ayke/pkg/mod/golang.org/x/tools@v0.5.1-0.20230114154351-e035d0c426c8/go/types/typeutil/map.go:86 +0x84
  github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:371 +0x44
  github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:438 +0x684
  github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:376 +0x94
  github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction()
      /home/ayke/src/tinygo/tinygo/compiler/symbol.go:68 +0x1c8
  github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:1799 +0x9c4
  github.com/tinygo-org/tinygo/compiler.(*builder).createExpr()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:1963 +0x618
  github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:1440 +0xa5c
  github.com/tinygo-org/tinygo/compiler.(*builder).createFunction()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:1338 +0xb88
  github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:854 +0x320
  github.com/tinygo-org/tinygo/compiler.CompilePackage()
      /home/ayke/src/tinygo/tinygo/compiler/compiler.go:307 +0x664
  github.com/tinygo-org/tinygo/builder.Build.func3()
      /home/ayke/src/tinygo/tinygo/builder/build.go:357 +0x254
  github.com/tinygo-org/tinygo/builder.runJob()
      /home/ayke/src/tinygo/tinygo/builder/jobs.go:222 +0x70
  github.com/tinygo-org/tinygo/builder.runJobs.func2()
      /home/ayke/src/tinygo/tinygo/builder/jobs.go:123 +0x40

Goroutine 44 (running) created at:
  github.com/tinygo-org/tinygo/builder.runJobs()
      /home/ayke/src/tinygo/tinygo/builder/jobs.go:123 +0x5fc
  github.com/tinygo-org/tinygo/builder.Build()
      /home/ayke/src/tinygo/tinygo/builder/build.go:877 +0x4e7c
  main.Build()
      /home/ayke/src/tinygo/tinygo/main.go:168 +0x1c4
  fmt.Sscanf()
      /usr/local/go/src/fmt/scan.go:114 +0x304
  github.com/tinygo-org/tinygo/goenv.GetGorootVersion()
      /home/ayke/src/tinygo/tinygo/goenv/version.go:42 +0x260
  github.com/tinygo-org/tinygo/builder.NewConfig()
      /home/ayke/src/tinygo/tinygo/builder/config.go:32 +0x154
  main.Build()
      /home/ayke/src/tinygo/tinygo/main.go:144 +0x44
  main.main()
      /home/ayke/src/tinygo/tinygo/main.go:1534 +0x447c

Goroutine 45 (running) created at:
  github.com/tinygo-org/tinygo/builder.runJobs()
      /home/ayke/src/tinygo/tinygo/builder/jobs.go:123 +0x5fc
  github.com/tinygo-org/tinygo/builder.Build()
      /home/ayke/src/tinygo/tinygo/builder/build.go:877 +0x4e7c
  main.Build()
      /home/ayke/src/tinygo/tinygo/main.go:168 +0x1c4
  fmt.Sscanf()
      /usr/local/go/src/fmt/scan.go:114 +0x304
  github.com/tinygo-org/tinygo/goenv.GetGorootVersion()
      /home/ayke/src/tinygo/tinygo/goenv/version.go:42 +0x260
  github.com/tinygo-org/tinygo/builder.NewConfig()
      /home/ayke/src/tinygo/tinygo/builder/config.go:32 +0x154
  main.Build()
      /home/ayke/src/tinygo/tinygo/main.go:144 +0x44
  main.main()
      /home/ayke/src/tinygo/tinygo/main.go:1534 +0x447c
==================
```

I reproduced it like this (in the fish shell):

```
$ go install -race
$ for i in (seq 1 50); rm ~/.cache/tinygo/pkg-*; tinygo build -o test -size=short -target=cortex-m-qemu testdata/generics.go; end
```

aykevl, 2023-03-08:
The race condition appears to be triggered when a package with a generic function is imported into two different packages and each package instantiates the generic function.
I assume there is some sort of cache that isn't properly locked.

aykevl, 2023-03-08:
I suspect this is a bug in golang.org/x/tools. I've verified the bug is still present after updating the package.

aykevl, 2023-03-08:
Verified that it is indeed just two packages racing with each other.

testa and testb:

```go
package testa

import (
        "github.com/tinygo-org/tinygo/testdata/generics/value"
)

func Test() {
        value.New(1)
}
```

value:

```go
package value

func New[T any](v T) {
}
```

When I delete the cached bitcode for testa and testb, it indeed logs a race condition sometimes.

dgryski, 2023-03-30:
So is this an upstream issue we need to work around or just wait for a fix or .. ?

aykevl, 2023-06-01:
Possibly related: https://github.com/golang/go/issues/59427

aykevl, 2023-06-07:
Probably the same issue as https://github.com/tinygo-org/tinygo/issues/3215.

deadprogram, 2023-09-21:
This is addressed as part of `v0.30.0` release so now closing. Thanks!

closed 2023-09-21 by deadprogram