tinygo-org/tinygo #3535: cannot initiate I2C connection – atsamd21e18a

hugolgst, 2023-03-09:
Hello,

I am trying to initiate an I2C connection with my custom board following the documentation. I can blink the LED and monitor the serial output to see my logs.

However, when I start adding the I2C code, there are errors:

1. With SERCOM1 or SERCOM3:
   I can flash the device and see my LED blinking; however, I get an error in my logs: "invalid clock pin," which is, I believe, coming from [this part of the code](https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/src/machine/machine_atsamd21.go#L685-L691).

2. With SERCOM0 or SERCOM2:
   When flashing the device, it apparently succeeds, but nothing is happening on my board - no LED flashing and no serial output to monitor. It is not even appearing. Using GDB, I can get the information that there has been a HardFault.

Here are some pieces of information about the board I am trying to add. It uses ATSAMD21E18A, which is already implemented. [This is the PR that adds my board](https://github.com/tinygo-org/tinygo/pull/3527) - CryptoAuth Trust Platform.

I have been trying for two weeks now to make my board work with TinyGo without success and am a bit desperate right now. Thanks in advance for your help.

Some additional information:
- SAMD21 I2C initialization - https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42631-SAM-D21-SERCOM-I2C-Configuration_ApplicationNote_AT11628.pdf
- CryptoAuth Trust Platform - https://www.microchip.com/en-us/development-tool/DM320118
- CryptoAuth Trust Platform Schematics - https://ww1.microchip.com/downloads/aemDocuments/documents/SCBU/ProductDocuments/BoardDesignFiles/DM320118-schematics.PDF

–hugo

deadprogram, 2023-03-09:
I made a few comments about errors in your board file. Hopefully that will help get you unstuck. :smile_cat:

aykevl, 2023-05-03:
@hugolgst have you tried looking into which SERCOMs are used for I2C (and possibly other peripherals) in other board support packages? You need to use the right SERCOM and can't re-use them for both I2C and UART for example (so if you use SERCOM0 for UART, you can't use it again for I2C).

deadprogram, 2024-05-12:
CLosing due to age, please reopen if needed. Thank you everyone!