tinygo-org/tinygo #3544: Want to easily modify USB_PID, etc.

sago35, 2023-03-13:
I would like to modify the following four values at tinygo build time.

* usb_STRING_PRODUCT
* usb_STRING_MANUFACTURER
* usb_VID
* usb_PID

https://github.com/tinygo-org/tinygo/blob/v0.27.0/src/machine/board_xiao-rp2040.go#L99-L108

I tried changing usbn_STRING_PRODUCT from const to var and rewriting it using ldflags, but it did not work.

```
$ tinygo flash --target xiao-rp2040 --ldflags "-X machine.usbn_STRING_PRODUCT='XXX'" examples/blinky1
```

ldflags can rewrite the strings contained in the main package.
However, for some reason, I think it cannot rewrite the machine package.

One way to resolve this Issue is to

1. Make ldflags work with any package that includes the machine package
2. Change machine.usb_STRING_PRODUCT etc. from const to var
3. Prepare string variables to override usb_VID and usb_PID

sago35, 2023-03-13:
I am making my keyboard in TinyGo.
I use xiao-rp2040 for my keyboard.

In this situation, if you want to use the xiao-rp2040 separately from the keyboard, the tinygo flash command becomes a bit cumbersome.
For example, if -port is specified incorrectly, the keyboard will not work, which is very annoying.

https://github.com/sago35/tinygo-keyboard

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you everyone!

closed 2023-06-14 by deadprogram