tinygo-org/tinygo #3572: Make https://github.com/ivmai/bdwgc available via command-line flag

dgryski, 2023-03-17:
For WASI, this is available via https://github.com/wasilibs/nottinygc

We should see what is required to make this work on native platforms, and make it available with `-gc=bdwgc`.

dgryski, 2023-03-17:
/cc @anuraaga

anuraaga, 2023-03-18:
I had looked into this a little but was struggling to get bdwgc to build against the macos minimal sdk that TinyGo uses for native build on Mac. It looked quite tricky so wasn't able to go further on it.

aykevl, 2023-05-03:
@anuraaga what was the error you were getting on macOS? This might be relatively easy to fix in macos-minimal-sdk.

anuraaga, 2023-05-08:
My very old branch is here

https://github.com/anuraaga/tinygo/tree/bdwgc

To start, I wasn't sure what flags to pass to get the command to use the minimal SDK when building on mac. This is probably something that could have been figured out relatively straightforwardly. With that branch, `make bdwgc` works but naturally the library can't be linked since it's not using minimal SDK.

But I thought it might be easier to get cgo to build it without using make - I was able to get the bdwgc code wired through cgo to be built, and then there were compilation errors related to missing header files IIRC. But there were other issues with the cgo approach that made me abandon it so don't remember the error exactly, and never went back to trying to figure out the flags to make to get it to use minimal SDK, and then probably have the same header file related error.

aykevl, 2023-05-08:
> To start, I wasn't sure what flags to pass to get the command to use the minimal SDK when building on mac.

None. It is always used when building binaries for Mac (TinyGo does not differentiate between compiling for the native OS or cross compiling, it doesn't use any headers/libraries from the host).

If you can try again and post the errors here that would be useful.

> But I thought it might be easier to get cgo to build it without using make

This would certainly be preferable, in fact I'm not sure I'd want to include bdwgc in TinyGo if it required building the library during a release (all libraries like picolibc, musl, etc are shipped as source code, with the notable exception of wasi-libc which is limited to a single architecture).

anuraaga, 2023-05-08:
The big problem with cgo and submodule was that, at least given the go documentation and issue tracker, there didn't seem to be any way to  compile c source without a go file in the same directory. I might have missed some technique though so if it's possible that would be great.

It's been a few months and I'm not in a position to help with native compilation of bdwgc right now (I wish I kept some of the code that creates error messages but don't seem to have pushed any, sorry...) but if anyone takes it up, look forward to archiving nottinygc!

aykevl, 2023-05-19:
For those who would like to implement this: don't use CGo - at least not for compiling the C files. Instead, look at how picolibc, musl, and compiler-rt are compiled. We need something similar for bdwgc.

davecheney, 2026-10-27:
Boehm GC is now available through `-gc=boehm`, including on native platforms. Thanks to @anuraaga for the early investigation, @aykevl for bringing the collector into TinyGo, @jakebailey for the subsequent improvements, and everyone else who contributed.

I verified this sample on macOS arm64 with TinyGo 0.44-dev (`b7ec945f`):

```go
package main

import "runtime"

var buffers [][]byte

func main() {
	for i := 0; i < 100; i++ {
		b := make([]byte, 1024)
		b[0] = byte(i)
		buffers = append(buffers, b)
	}
	runtime.GC()
	for i, b := range buffers {
		if b[0] != byte(i) {
			panic("retained allocation corrupted")
		}
	}
	println("Boehm GC allocation check passed")
}
```

```text
$ tinygo run -gc=boehm main.go
Boehm GC allocation check passed
```

Closing as completed.

closed 2026-10-27 by davecheney