tinygo-org/tinygo #3581: troubleshooting: fixed heap overhead for cgo invocation

anuraaga, 2023-03-20:
tl;dr Is there any known memory overhead when cgo is part of the AST that manifests at package initialization?

I am looking into heap memory usage in coraza-proxy-wasm. We use nottinygc, so malloc is implemented separately from the Go heap by mimalloc, and the Go heap is implemented by bdwgc.

I understand the slotting of these libraries complicates things compared to vanilla a lot, but wondering if anything intuitive comes to mind just based on symptoms.

The code I am looking at specifically is this cgo call

https://github.com/wasilibs/go-re2/blob/main/internal/cre2/cre2.go#L34

If I stub it out in a constant way, memory usage is low

if true {
  return nil
}

`Sys: 16646144, HeapSys: 10354688`

Sys is the total allocated memory, the sum of mimalloc (C heap) and bdwgc (Go heap) stats. HeapSys is just the Go heap reported by bdwgc.

If I stub out in a non-constant way, the go heap size blows up

if i != 0 {
  return nil
}
i++
(i is a package variable initialized to 0)

`Sys: 39124992, HeapSys: 32833536`

And without stubbing, e.g. compiling all of about 50 regexes rather than just 1,

`Sys: 43319296, HeapSys: 32833536`

The Go Heap is constant while C heap goes up.

Instrumenting runtime.alloc, I can confirm this codepath does not call it, as expected.

So I'm suspecting that when stubbed out in a constant way, something is removed completely from the AST that is related to the Go heap usage. As this codepath itself is not allocating, I wonder if it is something added to binary initialization itself.

Is there any known memory overhead when cgo is part of the AST that manifests at package initialization?

aykevl, 2023-05-03:
> Is there any known memory overhead when cgo is part of the AST that manifests at package initialization?

No, but there could be many things that trigger an escape to the heap. I suggest adding `-print-allocs=.` before and after the change to see whether any new allocations were added in your program.

davecheney, 2026-10-26:
Could you try this A/B comparison using the same target, scheduler, GC, and optimization settings for both builds? It keeps the Go caller identical and changes only a small helper from Go to C, measuring memory at entry to main and after repeated calls.

`main.go`:

```go
package main

import "runtime"

func main() {
	var before, after runtime.MemStats
	runtime.ReadMemStats(&before)

	var result uint32
	for i := uint32(0); i < 10000; i++ {
		result = step(result)
	}

	runtime.ReadMemStats(&after)

	println("result:", result)
	println("startup Sys:", before.Sys)
	println("startup HeapSys:", before.HeapSys)
	println("startup HeapAlloc:", before.HeapAlloc)
	println("call TotalAlloc:", after.TotalAlloc-before.TotalAlloc)
	println("call Mallocs:", after.Mallocs-before.Mallocs)
}
```

First build with this `step.go`:

```go
package main

//go:noinline
func step(x uint32) uint32 {
	return x + 1
}
```

Then replace `step.go` with:

```go
package main

/*
#include <stdint.h>

__attribute__((noinline))
uint32_t step_c(uint32_t x) {
	return x + 1;
}
*/
import "C"

func step(x uint32) uint32 {
	return uint32(C.step_c(C.uint32_t(x)))
}
```

Build the package directory (`tinygo build .`), rather than listing the two source files. Include `-print-allocs=.` and keep all other build settings identical between variants. Both should print `result: 10000`. Please share the output and allocation diagnostics for each variant, along with the build commands and TinyGo version.

This is an untested baseline, not a reproduction of the original allocator combination. It deliberately excludes C allocations, pointer passing, and library initialization. In particular, the ReadMemStats results depend on the selected allocator; they should not be assumed to include a separate C heap. A comparison using the original mimalloc/bdwgc configuration would also need its C-heap statistics.

anuraaga, 2026-10-26:
Sorry for still having this open, I haven't used TinyGo for a long time so can't debug this. There's a reasonable chance this was already fixed by now

closed 2026-10-26 by anuraaga

davecheney, 2026-10-26:
Not a problem, sorry for the delay. Please feel free to open a new issue if/when you reach for tinygo in the future.