tinygo-org/tinygo #3585: Memory usage differences between compling with `custommalloc`(but no custom malloc provided) and w/o it.

MengjiaLiang, 2023-03-21:
Hi there, I am doing some perf testing on a wasm plugin I wrote for Istio.
An interesting thing from my observation is the huge difference in memory usage of the Envoy proxy when compiling my plugin with `-tags=custommalloc` and without it, though I didn't provide my customized malloc.

Testing environment & tools:
Minikube: k8s v1.22 node
Istio(both v1.14.1 & v1.16.1)
TinyGo version: v0.27.0
Golang Version: 1.20.2

My wasm plugin is doing some regex parsing.

With QPS 600/s, 20K requests have been sent in total.
The memory usage of istio pod will be `560Mi` with the wasm plugin complied by
```
tinygo build -o plugin.wasm -gc=precise -target=wasi -scheduler=none ./regex/main.go
```

And it will only be `122Mi` with the wasm plugin compiled by
```
tinygo build -o plugin.wasm -gc=precise -tags=custommalloc -target=wasi -scheduler=none ./regex/main.go
```
Note: I didn't provide any customized malloc in my project.

I am curious that which mallow will be used when I give `-tags=custommalloc` but did not define my own malloc?

Thanks in advance for the help!

aykevl, 2023-03-21:
Are you _sure_ it doesn't include a custom GC somewhere? For example, by including https://github.com/wasilibs/nottinygc somewhere in one of the dependencies. Because I find it hard to believe it wouldn't give a linker error without a custom GC.

As to why the memory usage is different, you might be better off asking the maker of the custommalloc implementation.

MengjiaLiang, 2023-03-22:
Hi Ayke, thanks for the response!
You are right, I just checked my code. I have a few sub-directories under the project. One of the main.go file in the other directory imports the `nottinygc`, so the custommalloc must be from there.

I am closing this issue now. Thanks again.