tinygo-org/tinygo #3592: Darwin: Figure out how to make crypto/x509 happy

dgryski, 2023-03-23:
```
/Users/dgryski/go/src/go.googlesource.com/go/src/debug/elf/file_test.go:906:10: undefined: net.ResolveIPAddr
tinygo:ld.lld: error: undefined symbol: _crypto/x509/internal/macos.syscall
>>> referenced by corefoundation.go:77 (/Users/dgryski/go/src/go.googlesource.com/go/src/crypto/x509/internal/macos/corefoundation.go:77)
>>>               /Users/dgryski/Library/Caches/tinygo/thinlto/llvmcache-AD51E514D852B8C17CCBECA235FCF08F2E037423:(symbol _crypto/x509/internal/macos.StringToCFString+0x35)

tinygo:ld.lld: error: undefined symbol: _internal/abi.FuncPCABI0
>>> referenced by corefoundation.go:77 (/Users/dgryski/go/src/go.googlesource.com/go/src/crypto/x509/internal/macos/corefoundation.go:77)
>>>               /Users/dgryski/Library/Caches/tinygo/thinlto/llvmcache-AD51E514D852B8C17CCBECA235FCF08F2E037423:(symbol _crypto/x509/internal/macos.StringToCFString+0x19)

tinygo:ld.lld: error: undefined symbol: _crypto/x509/internal/macos.x509_CFStringCreateWithBytes_trampoline
>>> referenced by /<unknown>
>>>               /Users/dgryski/Library/Caches/tinygo/thinlto/llvmcache-AD51E514D852B8C17CCBECA235FCF08F2E037423:(symbol _crypto/x509/internal/macos$pack.6+0x8)
failed to run tool: ld.lld
error: failed to link /var/folders/5b/_hr1d1fd3qn4t9vtfx5p61wm0000gp/T/tinygo2564788547/main: exit status 1
```

Making testing stuff tricky.

deadprogram, 2023-08-18:
@dgryski it this still an issue on the latest `dev` branch?

davecheney, 2026-10-27:
The reported linker errors no longer reproduce with TinyGo 0.44-dev (`b7ec945f`) on macOS arm64. This sample compiles and runs:

```go
package main

import "crypto/x509"

func main() {
	roots, err := x509.SystemCertPool()
	if err != nil {
		panic(err)
	}
	if roots == nil {
		panic("nil system certificate pool")
	}
	println("SystemCertPool succeeded")
}
```

```text
$ tinygo run main.go
SystemCertPool succeeded
```

TinyGo’s macOS certificate APIs remain stubbed, so this confirms the linking problem is resolved, not that native system certificate verification is supported. Closing as completed for the reported linker failure.

closed 2026-10-27 by davecheney