tinygo-org/tinygo #3594: Crash when compiling cgo on Windows

anuraaga, 2023-03-24:
Certain patterns with cgo seem to result in a crash during compilation only on Windows. In this case it was the addition of nottinygc to a repo, so accessing bdwgc through cgo presumably.

https://github.com/wasilibs/go-re2/actions/runs/4508061182/jobs/7936426948?pr=34

```
Exception 0xc0000005 0x8 0x0 0x0
PC=0x0
signal arrived during external code execution

runtime.cgocall(0x7ff705546a50, 0xc00074a460)
	C:/hostedtoolcache/windows/go/1.20.0/x64/src/runtime/cgocall.go:157 +0x4a fp=0xc00074a438 sp=0xc00074a400 pc=0x7ff7051b4daa
github.com/tinygo-org/tinygo/cgo._Cfunc_clang_parseTranslationUnit2(0x20e1fb91880, 0x20e1fbb1080, 0x20e1fb8e510, 0xc, 0xc0020abfb0, 0x1, 0x1, 0xc0020d9748)
	_cgo_gotypes.go:860 +0x56 fp=0xc00074a460 sp=0xc00074a438 pc=0x7ff705470f96
github.com/tinygo-org/tinygo/cgo.(*cgoFile).readNames.func6(0xc000881bb0?, 0x47?, 0xc0020abc80?, {0xc001ec2780?, 0xc, 0x1?}, 0x0?, 0x7ff709af7e59?)
	D:/a/tinygo/tinygo/cgo/libclang.go:131 +0xfb fp=0xc00074a4d8 sp=0xc00074a460 pc=0x7ff70547383b
github.com/tinygo-org/tinygo/cgo.(*cgoFile).readNames(0xc0020ea600, {0xc0000ebb00, 0x43f}, {0xc001ec2780?, 0xc, 0x14}, {0xc0020cc6e1, 0x7}, 0xc00074aa28)
	D:/a/tinygo/tinygo/cgo/libclang.go:131 +0x3a5 fp=0xc00074a828 sp=0xc00074a4d8 pc=0x7ff70547[260](https://github.com/wasilibs/go-re2/actions/runs/4508061182/jobs/7936426948?pr=34#step:8:261)5
github.com/tinygo-org/tinygo/cgo.Process({0xc001fbf0f8, 0x1, 0x40?}, {0xc00042a5d0, 0x12}, {0xc000396060, 0x28}, 0xc0002fc2c0, {0xc0020a0800, 0x9, ...}, ...)
	D:/a/tinygo/tinygo/cgo/cgo.go:320 +0xced fp=0xc00074ab48 sp=0xc00074a828 pc=0x7ff7054652ed
github.com/tinygo-org/tinygo/loader.(*Package).parseFiles(0xc00014b860)
	D:/a/tinygo/tinygo/loader/loader.go:443 +0x36e fp=0xc00074ad88 sp=0xc00074ab48 pc=0x7ff705483e4e
github.com/tinygo-org/tinygo/loader.(*Package).Parse(0xc00014b860)
	D:/a/tinygo/tinygo/loader/loader.go:358 +0x7f fp=0xc00074ada0 sp=0xc00074ad88 pc=0x7ff70548353f
github.com/tinygo-org/tinygo/loader.(*Program).Parse(0xc000097380)
	D:/a/tinygo/tinygo/loader/loader.go:307 +0x65 fp=0xc00074add8 sp=0xc00074ada0 pc=0x7ff705483185
github.com/tinygo-org/tinygo/builder.Build({0xc00001fce0, 0x27}, {0x0, 0x0}, {0xc00039c1c0, 0x35}, 0xc000033980)
	D:/a/tinygo/tinygo/builder/build.go:205 +0x1307 fp=0xc00074b978 sp=0xc00074add8 pc=0x7ff7054f5647
main.buildAndRun({0xc00001fce0, 0x27}, 0xc000033980, {0x7ff709c16c60, 0xc0000339b0}, {0xc000038170, 0x1, 0x1}, {0x0, 0x0, ...}, ...)
	D:/a/tinygo/tinygo/main.go:827 +0x590 fp=0xc00074bc08 sp=0xc00074b978 pc=0x7ff70552e030
main.Test({0xc00001fce0, 0x27}, {0x7ff709c17620, 0xc0000490a8}, {0x0?, 0x670a6c616e7[265](https://github.com/wasilibs/go-re2/actions/runs/4508061182/jobs/7936426948?pr=34#step:8:266)74?}, 0x6f632e6[275](https://github.com/wasilibs/go-re2/actions/runs/4508061182/jobs/7936426948?pr=34#step:8:276)687469?, 0x0, 0x1, 0x0, ...)
	D:/a/tinygo/tinygo/main.go:250 +0x675 fp=0xc00074bea8 sp=0xc00074bc08 pc=0x7ff705528e75
main.main.func3()
	D:/a/tinygo/tinygo/main.go:1653 +0x24e fp=0xc00074bfe0 sp=0xc00074bea8 pc=0x7ff70553634e
runtime.goexit()
	C:/hostedtoolcache/windows/go/1.20.0/x64/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc00074bfe8 sp=0xc00074bfe0 pc=0x7ff705218861
created by main.main
	D:/a/tinygo/tinygo/main.go:1647 +0x35c9
```

davecheney, 2026-10-27:
Sorry for the long delay in following up. Does this still reproduce with a current TinyGo release? If so, could you share the TinyGo version and a minimal CGo example that triggers the crash on Windows?

anuraaga, 2026-10-27:
Not sure but likely not!

closed 2026-10-27 by anuraaga

davecheney, 2026-10-27:
>  Not sure but likely not!

np, please reopen this issue if you are able to share a program which reproduces the crash.